Kinovarobotics/kinova-ros #356: Problem with obstacles and collision objects

s265452, 2021-09-20:
Hi everyone,
I have some problems with my Jaco2 7DoF. 
When I launch a pick and place with obstacles in the environment, sometimes happens that the arm collide with obstacles, same as in the image attached.
I think this happens when physical robot reach joint limits and it's impossible to follow the path. In these conditions I lose the control of the arm.

Planning Library: OMPL
kinematics_solver: trac_ik_kinematics_plugin/TRAC_IKKinematicsPlugin
solve_type: Distance

How can I solve these problems??
[image]

felixmaisonneuve, 2021-09-20:
Hi @s265452,

Can you give the error messages you get when such problem occurs?

Also, you say this could be because the physical robot reach its joint limits. Joint limits are defined in xacro files, and I think it is these values that are used when doing a pick and place. It is possible that limits are not fully accurate with actual limits of the physical robot. In some extreme cases, I think it could be possible, when you launch a pick and place, that it tries to move the robot beyond its actual limits, which result in unexpected behaviour. 

One way to verify that would be to monitor joint limits and check if it gets close to limits defined in `kinova_description/urdf/<your_robot>.xacro`. If it seems to be the cause of your issue, maybe you could try to reduce those values. 

Regards,
Felix

s265452, 2021-09-25:
Hi @felixmaisonneuve,
Thank you for yout prompt reply.

I did a Check on Joint limits and I don't found nothing strange.

I read the Warnings in the image attached.

[image]

I don't know if the problem could be the IK solver.
I followed the instructions at this page: https://moveit.readthedocs.io/en/latest/doc/trac_ik_tutorial.html

sudo apt-get install ros-melodic-trac-ik

sudo apt-get install ros-melodic-trac-ik-kinematics-plugin

sudo apt-get install ros-melodic-trac-ik-lib

But I didn't add to my catkin ws. 

Waiting for your answer.
Regards.

felixmaisonneuve, 2021-09-27:
I compared joint limits in the URDF file and the ones in the User Guide and, like you said, they are identical.

When I try our `pick_and_place` example I also get these warnings. I wouldn't try to get rid of them yet. 
You do need to install `ros-melodic-trac-ik`, `ros-melodic-trac-ik-kinematics-plugin` and `ros-melodic-moveit`, but since it seems to work, I think these packages are already installed on your system. 

Just to clarify, from what I understand, your `pick_and_place_1` example works most of the time. Only in "extreme" cases your robot will collide with an obstacle and you will lose control of the arm.

Does the arm always collide with the same objects? Or does it collide with any object?
Is this frequent? Or does it only happens rarely in some more complex cases? 
If you simplify your environnement (using less complex objects, giving more space to move to the arm..), does it solve the issue? In other words, does this happen because the arm has no solution to reach the goal and it goes crazy?
Have you tried using simple object shapes (like a box or a sphere) to see if it is your objects that are not correctly computed in the IK Solver?
If you think this happens when joints reach their limits, I would recommend to reduce the values in `kinova_description/urdf/j2s7s300.xacro` (just one or two degrees should be enough e.g. changing `${47/180*J_PI}` to `${45/180*J_PI}`). You will need to change upper and lower limit for each `revolute` joint (joints 2,4 and 6 in your case). Maybe the float rounding is enough to cause this error (but I highly doubt it). 

Finally, when the arm collides with an object, are you sure there is no error message in either your `roslaunch j2s7s300_moveit_config ...` terminal or your `rosrun kinova_arm_moveit_demo ...` one?

Regards,
Felix

closed 2021-10-06 by s265452