Kinovarobotics/kinova-ros #364: I have a question about the topic:"/j2n6s200_driver/trajectory_controller/command"

xia3613389, 2021-11-12:
## Description

I want to use the topic /j2n6s200_driver/trajectory_controller/command to control my kinova robot(j2n6s200), I don't know much about this topic.So I first used the move_robot.py(kinova_control) program to run in gazebo,It runs normally without any problems.
Then ,I use the topic /j2n6s200_driver/trajectory_controller/command to publish a posetion,The result of robotic arm movement is inconsistent with the track point I posted.The robot arm moves very fast, and there is a danger of collision, so I can't continue to verify my code.

## Version

ROS distribution : noetic

Branch and commit you are using : 

## Steps to reproduce

1. *step 1...*  roslaunch kinova_bringup kinova_robot.launch

2. *step 2...*  rosrun kinova_control move_robot_test.py j2n6s200

3. result：
 [ INFO] [1636687712.693278112]: Trajectory controller Receive trajectory with points number: 1
 [ INFO] [1636687713.698124140]: Moved to point 0
current_position [275.4723205566406, 167.40382385253906, 57.582489013671875, 240.52496337890625, 82.90911102294922, 75.551025390625, 0.0]
goal_position [285.4723205566406, 167.40382385253906, 57.582489013671875, 240.52496337890625, 82.90911102294922, 75.551025390625, 0.0]
final_position [281.17010498046875, 173.10458374023438, 63.28257369995117, 245.68606567382812, 82.90911102294922, 75.551025390625, 0.0]

The final_position is not goal_position ，the result of robotic arm movement is inconsistent with the track point I posted.The log shows： Moved to point 0. I don't know if point 0 is the track point I posted or a default track point, because I can't let it run to this point, there is a danger of collision.If it is the trajectory point I released, then it should follow the normal trajectory. So I am very confused.

## Code example (if necessary)
this is my code to use topic /j2n6s200_driver/trajectory_controller/command
move_robot_test.py
```
    #! /usr/bin/env python3
"""Publishes joint trajectory to move robot to given pose"""

import rospy
from trajectory_msgs.msg import JointTrajectory
from trajectory_msgs.msg import JointTrajectoryPoint
from std_srvs.srv import Empty
import argparse
import time
import kinova_msgs.msg
import actionlib
import std_msgs.msg
from kinova_msgs.srv import *
import geometry_msgs.msg
import sys

def argumentParser(argument):
  """ Argument parser """
  parser = argparse.ArgumentParser(description='Drive robot joint to command position')
  parser.add_argument('kinova_robotType', metavar='kinova_robotType', type=str, default='j2n6a300',
                    help='kinova_RobotType is in format of: [{j|m|r|c}{1|2}{s|n}{4|6|7}{s|a}{2|3}{0}{0}]. eg: j2n6a300 refers to jaco v2 6DOF assistive 3fingers. Please be noted that not all options are valided for different robot types.')
  #args_ = parser.parse_args(argument)
  argv = rospy.myargv()
  args_ = parser.parse_args(argv[1:])
  prefix = args_.kinova_robotType
  nbJoints = int(args_.kinova_robotType[3])	
  nbfingers = int(args_.kinova_robotType[5])	
  return prefix, nbJoints, nbfingers

def moveJoint (jointcmds,prefix,nbJoints):
  # topic_name = '/' + prefix + '/trajectory_controller/command'
  pub = rospy.Publisher("/j2n6s200_driver/trajectory_controller/command", JointTrajectory, queue_size=10)
  jointCmd = JointTrajectory()  
  point = JointTrajectoryPoint()
  jointCmd.header.stamp = rospy.Time.now() + rospy.Duration.from_sec(0.0);  
  point.time_from_start = rospy.Duration.from_sec(1.0)
  for i in range(0, nbJoints):
    jointCmd.joint_names.append(prefix +'_joint_'+str(i+1))
    point.positions.append(jointcmds[i])
    point.velocities.append(0) 
    point.accelerations.append(0)
    point.effort.append(0)
  # print(point)
  jointCmd.points.append(point)
  rate = rospy.Rate(10)
  count = 0
  while (count < 50):
    
    pub.publish(jointCmd)
    count = count + 1
    rate.sleep()

def getcurrentJointCommand():
    # wait to get current position
    topic_address = '/j2n6s200_driver/out/joint_command'
    rospy.Subscriber(topic_address, kinova_msgs.msg.JointAngles, setcurrentJointCommand)
    rospy.wait_for_message(topic_address, kinova_msgs.msg.JointAngles)
   
def setcurrentJointCommand(feedback):
    global currentJointCommand

    currentJointCommand_str_list = str(feedback).split("\n")
    for index in range(0,len(currentJointCommand_str_list)):
        temp_str=currentJointCommand_str_list[index].split(": ")
        currentJointCommand[index] = float(temp_str[1])

if __name__ == '__main__':
  rospy.init_node('move_robot_using_trajectory_msg')		
  prefix, nbJoints, nbfingers = argumentParser(None)  

  time.sleep(2)
  currentJointCommand=[0]*7

  getcurrentJointCommand()
  a_position=currentJointCommand[:]
  b_position=currentJointCommand[:]
  diff=10.0
  b_position[0]=b_position[0]+diff
  print("current_position",a_position[:])
  print("goal_position",b_position[:])

  start=rospy.ServiceProxy('/j2n6s200_driver/in/start',Start)
  start()                   
  try:    
     
    if (nbJoints==6):
      moveJoint(b_position[0:6],prefix,nbJoints)
      time.sleep(5)
      a_position=currentJointCommand[:]
      print("final_position",a_position[:])

  except rospy.ROSInterruptException:
    print("program interrupted before completion")

```

## Expected behavior

The result of robotic arm movement is inconsistent with the track point I posted when I use topic "/j2n6s200_driver/trajectory_controller/command"

## Any other information

Any other information you believe the maintainers need to know.

felixmaisonneuve, 2021-11-12:
Hi @xia3613389,

I do not think you will be able to use `/j2n6s200_driver/trajectory_controller/command` with a real arm. Topics are different when being in a simulation and when using a real arm. see #291 

You cannot use the same code for both simulation and a real arm.

What are you trying to do? I can help you move your arm, but it won't be by using `/j2n6s200_driver/trajectory_controller/command`.

Best,
Felix

xia3613389, 2021-11-12:
Hi Felix,

I am sorry to bother you.
Recently I want to publish some trajectory points to make my kinova robotic arm move according to the trajectory points. I saw that there is a simulation example code in the kinova_control function package. It can publish trajectory points to virtual kinova in gezebo. Use the /effort_joint_trajectory_controller/command topic to publish trajectory points, and the virtual arm can also move normally according to the trajectory points. In the topic of kinova, I found the topic /j2n6s200_driver/trajectory_controller/command. I thought the functions of the two are the same, except that one controls the virtual arm and the other controls the real arm. So I wrote my own program imitating the example code, using /j2n6s200_driver/trajectory_controller/command. Then it does not move as I imagined. I know so little about the topic /j2n6s200_driver/trajectory_controller/command, that I simply think it can control a real robotic arm.
I'm really sorry for taking up your time.My English is bad, I hope my expression does not cause you problems.
Best，
Xia

felixmaisonneuve, 2021-11-12:
You are right, you can use the topic `/j2n6s200_driver/trajectory_controller/command` to publish a trajectory to a real arm. I have read too fast and I thought you wanted to use the same topic (`/effort_joint_trajectory_controller/command`) for both simulation (Gazebo) and a real arm. 

`/j2n6s200_driver/trajectory_controller/command` takes a `trajetcory_msgs/JointTrajectory` as an input. You can see the structure of this object in the ROS Documentation [here](http://docs.ros.org/en/melodic/api/trajectory_msgs/html/msg/JointTrajectory.html). 

But looking at your code (which is inspired by our `move_robot.py`), I am not sur you want to create full trajectories (with multiple points).

If you want to move in Cartesian Space (xyz coordinates), you can follow [this](https://github.com/Kinovarobotics/kinova-ros#cartesian-position-control) section. 
You have 2 options : 
1. Using our [pose_action_client.py](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_demo/nodes/kinova_demo/pose_action_client.py) example, which sends a goal to topic `/j2n6s300_driver/pose_action/tool_pose/goal`
2. Publishing to topic `/j2n6s300_driver/in/cartesian_velocity`

If you want to move in Angular Space (controlling each joint individually), you can follow [this](https://github.com/Kinovarobotics/kinova-ros#joint-position-control) section. 
In the same way, you can 
1. Use our [joints_action_client.py](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_demo/nodes/kinova_demo/joints_action_client.py) example that publishes goals at `/j2n6s300_driver/joints_action/joint_angles/goal`
2. publish on topic `/j2n6s300_driver/in/joint_velocity`

I think those examples probably better fit want you are trying to achieve, but please correct me if I'm wrong.
`/j2n6s200_driver/trajectory_controller/command` is usefull if you want to send a full trajectory (with multiple points), but looking at your example it seems like you are only trying to reach a single position. For that, I think it will be easier for you to use our other examples.

Regards,
Felix

xia3613389, 2021-11-13:
I read your suggestions carefully and read the routines you mentioned.That is very helpful to me.

I want to use the topic /j2n6s200_driver/trajectory_controller/command to publish a trajectorie, so I first wrote a program(move_joint_test.py) imitating a routine. Publish several trajectory points in the program. To be precise, these trajectory points should only make joint 1 move. But as show in the video, the motion of the robotic arm is not what I imagined. Its other joints also moved, and joint 1 did not move correctly.I am guessing whether the trajectory points I released are problematic, because I did not specify their speed, acceleration, or force. I am not sure if they are necessary. Or I made some errors elsewhere, but I couldn't find them.

Even if I only release one track point, the robotic arm should move to this point smoothly. But the result is not like that. The posted track points are inconsistent with the results of the motion, this is where I am questioning. There must be something wrong with me.I hope you can give me some suggestions to solve it.

Regards,

 Xia

felixmaisonneuve, 2021-11-15:
So I have tried your code and experimented a bit. I wasn't able to make the arm move at all. I do not understand how you managed to move your arm.

I think this topic is used by MoveIt. Moveit probably computes a full trajectory with velocities and accelerations, so it is possible your trajectory does not work because it is missing elements, but then again, you managed to move your arm without these.

Did you add something else to your code? Since I am unable to move the arm using your demo, it is hard for me to find what the issue is.

Once again, I would recommend looking into the much easier `/j2n6s200_driver/joints_action/joint_angles` topic, which is used in our [joints_action_client.py](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_demo/nodes/kinova_demo/joints_action_client.py). 

Regards,
Felix

xia3613389, 2021-11-18:
I did not add other elements to the robotic arm, I only set the position information of the track points. But I can use these track points to make my arm move, although the result of the exercise is wrong. 

I think you are right. This topic requires that the trajectory has complete position, velocity and acceleration information to ensure that the manipulator can move correctly.I will also seriously consider your suggestions. I think that next I should try to use moveit to plan these trajectories, I need to learn more  about it. 

Thank you very much for taking the time to answer my questions.Best wishes to you.

Regards,
Xia

felixmaisonneuve, 2021-11-18:
I think you will have more success using MoveIt and the other alternatives I suggested. Our readme files contain a lot of information on how to get you started. Make sure to go throught them.

I will close this issue, but you can always re-open it or open a new one if you have any question.

Regards,
Felix

AndrejOrsula, 2021-12-16:
Apologies @felixmaisonneuve for resurrecting this issue, but I believe @xia3613389 is correct. There seems to be some problem with publishing certain `JointTrajectory` messages directly to `${ROBOT_TYPE}_driver/trajectory_controller/command`.

I am trying to use [moveit_servo](https://moveit.ros.org/moveit/ros2/servo/jog/2020/09/09/moveit2-servo.html) to control `js7s300` in real-time using Cartesian commands (with collision avoidance). MoveIt Servo publishes short-term `JointTrajectory` messages, which I plugged into `${ROBOT_TYPE}_driver/trajectory_controller/command`. However, I am experiencing the same issue as @xia3613389 described above.
> The result of robotic arm movement is inconsistent with the track point I posted. The robot arm moves very fast, and there is a danger of collision, so I can't continue to verify my code.

I.e. the robot moves in a completely different direction and it does not respect the provided velocity/position command. Similarly, it is difficult to debug properly due to fast motion of the robot. As far as I can tell, the action motion is always in the same general direction regardless of what input I provide.

I am certain that my `kinova` and `moveit2` setup are correct as they work using fake controllers and simulation. I should also point out that kinematic paths planned by MoveIt2 work fine on the real robot too (using `/${ROBOT_NAME}/follow_joint_trajectory` action server via [joint_trajectory_action_server.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_driver/src/joint_trajectory_action/joint_trajectory_action_server.cpp)).

Looking closer, [joint_trajectory_action_server.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_driver/src/joint_trajectory_action/joint_trajectory_action_server.cpp) actually just republishes the received trajectory as `JointTrajectory` message under `${ROBOT_TYPE}_driver/trajectory_controller/command` so that [kinova_joint_trajectory_controller.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_driver/src/kinova_joint_trajectory_controller.cpp) can subscribe to it. The difference between using `/${ROBOT_NAME}/follow_joint_trajectory` action and `${ROBOT_TYPE}_driver/trajectory_controller/command` is therefore only in the processing done inside [joint_trajectory_action_server.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_driver/src/joint_trajectory_action/joint_trajectory_action_server.cpp), but I cannot see how it would change anything. The other potential difference is in the `JointTrajectory` message itself, which from `moveit_servo` often contains only a single point.  However, joint trajectory controller should support all trajectories, including those that contain only a single point.

Based on everything I described above, I am thinking there might be some **logical issue inside [kinova_joint_trajectory_controller.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_driver/src/kinova_joint_trajectory_controller.cpp)**? Maybe the point commands are not written for the 1st point or something and it's using whatever values are originally stored in certain variable(s) during initialisation, e.g. whatever garbage was in the memory before. It is just a speculation, as I am having troubles debugging it due to the fast random motion of the robot.

---

To provide some more information, I am including examples of important data messages.

```bash
# State of the robot before sending `JointTrajectory` message from `moveit_servo` to the controller
❯ ros2 topic type /joint_states
sensor_msgs/msg/JointState
❯ ros2 topic echo /joint_states
header:
  stamp:
    sec: 1639665710
    nanosec: 84462792
  frame_id: ''
name:
- robot_j2s7s300_joint_1
- robot_j2s7s300_joint_2
- robot_j2s7s300_joint_3
- robot_j2s7s300_joint_4
- robot_j2s7s300_joint_5
- robot_j2s7s300_joint_6
- robot_j2s7s300_joint_7
position:
- 0.0018084276467364747
- 2.683118998977502
- -0.0033735236282760187
- 4.121928174180132
- 0.0013834015359458175
- 1.4386929372307136
- -0.0011059532909349508
velocity:
- ~0.0
- ~0.0
- ~0.0
- ~0.0
- ~0.0
- ~0.0
- ~0.0
effort:
...
```

```bash
# Input into `moveit_servo` (e.g. user-given via teleop)
❯ ros2 topic type /delta_twist_cmds 
geometry_msgs/msg/TwistStamped
❯ ros2 topic echo /delta_twist_cmds
header:
  stamp:
    sec: 1639665710
    nanosec: 80913347
  frame_id: robot_j2s7s300_link_base
twist:
  linear:
    x: 0.01
    y: 0.0
    z: 0.0
[... 26 lines cut ...]
  - 0.0018269936084582235
  - 2.6513839135541986
  - -0.0036270128097474746
  - 4.070310504329728
  - 0.0016111866350521993
  - 1.4188099526732234
  - -0.0011914739106521231
  velocities:
  - 0.0000122039104886746
  - -0.020860332885570543
  - -0.0001666253182511495
  - -0.03392969520939815
  - 0.0001497293273470309
  - -0.013069625340391688
  - -0.0000562150242258523
  accelerations: []
  effort: []
  time_from_start:
    sec: 0
    nanosec: 50000000
```

Since the robot moves in a completely different direction, the subsequently observed `sensor_msgs/JointState` on ` /joint_states` are moving away from the target `JointTrajectoryPoint` posted above.

Note that `header/stamp` is not accessed by [kinova_joint_trajectory_controller.cpp](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_driver/src/kinova_joint_trajectory_controller.cpp). Therefore, I do not believe it matters that it's not set (`0.0 s`).

felixmaisonneuve, 2021-12-17:
Hi @AndrejOrsula,

I wasn't able to reproduce this issue last time, but I may have gone fast. I have looked into it deeper and, like you said, I think this points to kinova_joint_trajectory_controller.cpp. 

There is some curious things in this file. For example, I noticed `pub_joint_velocity_` publishes using `pn` node handle as opposed to `nh_`. I am not sure why. 
https://github.com/Kinovarobotics/kinova-ros/blob/6b7a12199b29dc7c9ee484eb890bd89e9cf20cd9/kinova_driver/src/kinova_joint_trajectory_controller.cpp#L27

I do not currently have access to an arm, but monday I will do some testing and investigate this issue.

Regards,
Felix

AndrejOrsula, 2021-12-17:
Thank you @felixmaisonneuve for looking into it.

One more thing I might add is that `time_from_start` of the point is respected (the random motion lasts exactly as long as the value in this field).

---

After a very brief look through the code, the issue might be at [kinova_joint_trajectory_controller.cpp#L175](https://github.com/Kinovarobotics/kinova-ros/blob/6b7a12199b29dc7c9ee484eb890bd89e9cf20cd9/kinova_driver/src/kinova_joint_trajectory_controller.cpp#L175) as I believe `traj_command_points_.size()-2` is outside of the bounds if the trajectory contains only a single point. Therefore, the only command contained in the message is being overwritten by some outside-of-bounds memory. I am not exactly why it doesn't result in a segmentation fault as it should, but  I have not investigated the data structures yet.

```cpp
    // replace last velocity command (which is zero) to previous non-zero value, trying to drive robot moving a forward to get closer to the goal.
    kinova_angle_command_[traj_command_points_.size()-1] = kinova_angle_command_[traj_command_points_.size()-2];
```

Change below could fix the current issue for a single-point trajectories, while preserving the current behaviour for all other trajectories. Although, I don't really understand why this exact line would ever be needed. Trajectories might or might not have `0.0` joint velocity command at the last point. Both scenarios are valid, so I don't see a reason to overwrite them. The file is over 5 years old now, so it is difficult to tell.

```diff
-     kinova_angle_command_[traj_command_points_.size()-1] = kinova_angle_command_[traj_command_points_.size()-2];
+     if (traj_command_points_.size() > 1)
+         kinova_angle_command_[traj_command_points_.size()-1] = kinova_angle_command_[traj_command_points_.size()-2];
```

I have not looked at the code as whole yet, so I could be totally wrong. I don't have access to the robot right now so I have not tested it.

AndrejOrsula, 2021-12-20:
@felixmaisonneuve Tested it on a real robot today and I can confirm that **removing [kinova_joint_trajectory_controller.cpp#L175](https://github.com/Kinovarobotics/kinova-ros/blob/6b7a12199b29dc7c9ee484eb890bd89e9cf20cd9/kinova_driver/src/kinova_joint_trajectory_controller.cpp#L175) fixes the issue** for me. It enables the use of `moveit_servo`, and I have not noticed any kind of regression for kinematic paths with multiple trajectory points.

```diff
-    // replace last velocity command (which is zero) to previous non-zero value, trying to drive robot moving a forward to get closer to the goal.
-    kinova_angle_command_[traj_command_points_.size()-1] = kinova_angle_command_[traj_command_points_.size()-2];
```

I recommend to simply remove this line, unless there is someone who knows of any reason to keep it. Should I open PRs?

closed 2021-12-20 by felixmaisonneuve