Kinovarobotics/kinova-ros #366: How to subtract the gravity torques of each joints of j2s6s300 to obtain the torque generated by the joint to overcome the external force？

Tzxtoiny, 2021-11-20:
I calculated the Jacobian matrix of the manipulator through DH parameters, and then wanted to get the Cartesian force  through the torque of each joint, but I found that the output each joint torque usually includes gravity torque. I want to know how to know the gravity torque under different configurations? 
When I checked the API document, I also found that the gravity torque needs to be reduced before admittance control, but I don't know how to obtain the gravity torque of the manipulator in different configurations. Please give me some inspiration, thank you.
!

[屏幕截图 2021-11-21 002238](https://user-images.githubusercontent.com/62111393/142733595-d491d80b-ab02-4c9d-9fec-c442cbe59f3b.png)

felixmaisonneuve, 2021-11-22:
Hi @Tzxtoiny,

There is 2 API calls that could help you : `GetAngularForceGravityFree` and `GetAngularTorqueGravityEstimation`.

I think `GetAngularForceGravityFree` will give you the current torque value of each joint without gravity. 

`GetAngularTorqueGravityEstimation` will give you the torque values created by gravity for each joint. I think this is what you are looking for.

In short, `joint_torque` = `GetAngularForceGravityFree` + `GetAngularTorqueGravityEstimation`.

To use those, you could create your own service that you could call at any given configuration. To do it, you would need to follow the steps I explained in #365. 

Note that `GetAngularTorqueGravityEstimation` already has a function in kinova_comm.cpp, you can call `KinovaComm::getGravityCompensatedTorques` directly. 

Best,
Felix

Tzxtoiny, 2021-11-23:
Thank you for your answer, which helped me a lot！！
Can you give me an example about callingKinovaComm::getGravityCompensatedTorques  function? I tried several times but failed.

Tzxtoiny, 2021-11-23:
> Hi @Tzxtoiny,
>
> There is 2 API calls that could help you : `GetAngularForceGravityFree` and `GetAngularTorqueGravityEstimation`.
>
> I think `GetAngularForceGravityFree` will give you the current torque [...]

Thank you for your answer, which helped me a lot！！
Can you give me an example about callingKinovaComm::getGravityCompensatedTorques function? I tried several times but failed.

Tzxtoiny, 2021-11-23:
> Hi @Tzxtoiny,
>
> There is 2 API calls that could help you : `GetAngularForceGravityFree` and `GetAngularTorqueGravityEstimation`.
>
> I think `GetAngularForceGravityFree` will give you the current torque [...]

Thank you for your answer, which helped me a lot！！
Can you give me an example about calling `KinovaComm::getGravityCompensatedTorques` function? I tried several times but failed.

felixmaisonneuve, 2021-11-24:
Hi @Tzxtoiny,

I can give you an example where I publish all three values on different topic that you can monitor. 

I first created my publishers, in kinova_arm.h
```c++
    ros::Publisher joint_torque_publisher_;
    ros::Publisher joint_torque_gravity_free_publisher_;    // new
    ros::Publisher joint_torque_gravity_estimation_publisher_;   // new
    ros::Publisher tool_wrench_publisher_;
    ros::Publisher finger_position_publisher_;
```

then in kinova_arm.cpp
```c++
    joint_torque_publisher_ = node_handle_.advertise<kinova_msgs::JointAngles>
            ("out/joint_torques", 2);
    joint_torque_gravity_free_publisher_ = node_handle_.advertise<kinova_msgs::JointAngles> // new
            ("out/joint_torques_gravity_free", 2);
    joint_torque_gravity_estimation_publisher_ = node_handle_.advertise<kinova_msgs::JointAngles> // new
            ("out/joint_torques_gravity_compensation", 2);
    joint_state_publisher_ = node_handle_.advertise<sensor_msgs::JointState>
            ("out/joint_state", 2);
```

In that same file, in the function `publishJointAngles`

```c++
    // Joint torques (effort)
    KinovaAngles joint_tqs;
    KinovaAngles joint_tqs_gravity_free;
    kinova_msgs::JointAngles joint_tqs_gravity_compensation;
    bool gravity_comp;
    // node_handle_.param("torque_parameters/publish_torque_with_gravity_compensation", gravity_comp, false);
    // if (gravity_comp==true)
    //   kinova_comm_.getGravityCompensatedTorques(joint_tqs);
    // else
    //   kinova_comm_.getJointTorques(joint_tqs);
    kinova_comm_.getJointTorques(joint_tqs);
    kinova_comm_.getGravityCompensatedTorques(joint_tqs_gravity_free);
    kinova_comm_.getTorqueGravityEstimation(gravity_compensation_);

    joint_tqs_gravity_compensation.joint1 = gravity_compensation_[0];
    joint_tqs_gravity_compensation.joint2 = gravity_compensation_[1];
    joint_tqs_gravity_compensation.joint3 = gravity_compensation_[2];
    joint_tqs_gravity_compensation.joint4 = gravity_compensation_[3];
    joint_tqs_gravity_compensation.joint5 = gravity_compensation_[4];
    joint_tqs_gravity_compensation.joint6 = gravity_compensation_[5];
    joint_tqs_gravity_compensation.joint7 = gravity_compensation_[6];

    joint_torque_publisher_.publish(joint_tqs.constructAnglesMsg());
    joint_torque_gravity_free_publisher_.publish(joint_tqs_gravity_free.constructAnglesMsg());
    joint_torque_gravity_estimation_publisher_.publish(joint_tqs_gravity_compensation);
```

For that, I created a new variable `gravity_compensation_` (because I needed an array for `getTorqueGravityEstimation`) that I declared in kinova_arm.h
```c++
    float l_joint_torque_[COMMAND_SIZE];
    float l_force_cmd_[COMMAND_SIZE];
    float gravity_compensation_[COMMAND_SIZE]; // here
```

and initialized in kinova_arm.cpp
```c++
    for (int i=0;i<COMMAND_SIZE;i++)
    {
      l_joint_torque_[i] = 0;
      l_force_cmd_[i] = 0;
      gravity_compensation_[i] = 0; // here
    }
```

After that, all three are published and you can echo them as you want
[image]

As we can see, `/out/joint_torques` is roughly equals to `/out/joint_torques_gravity_compensation` plus `out/joint_torques_gravity_compensation`.

Also, you can set the ROS param `torque_parameters/publish_torque_with_gravity_compensation` (manually or by changing its value in kinova_bringup/launch/config/robot_parameters.yaml before launching the driver), and the values provided by `KinovaComm::getGravityCompensatedTorques` will be published inthe topic `/out/joint_torques` without having to change anything (see the `if`/`else` I commented in `publishJointAngles`). 

Regards,
Felix

Tzxtoiny, 2021-11-24:
> Hi @Tzxtoiny,
>
> I can give you an example where I publish all three values on different topic that you can monitor.
>
> I first created my publishers, in kinova_arm.h
>
> ```c++
>     ros::Publisher [...]

@felixmaisonneuve Thank you very much for your example. I have benefited a lot！
I also found that the output torque can be achieved by modifying the parameters 'publish_torque_with_gravity_compensation: true' in the 'robot_parameters.yaml' file.  After modifying the parameters, I got the correct results, as shown in the screenshot you provided above. 
But I also found that the joint torque after gravity compensation is still not zero. Is this caused by friction, damping force or other forces? Is there any way to compensate these forces? （I want to calculate the external force at the end effector through Jacobi using the torque of each joint  to achieve admittance control. But the other forces may make it not so accuracy.)

felixmaisonneuve, 2021-11-24:
Hi @Tzxtoiny,

You can reset your torques using Development Center (in *Advanced Settings*->*Set Zero Torque*->*apply to all*) or the `in/set_zero_torques` service. This should reset your torques to 0. 

To do so, you first need to put your arm in a candle like position (all joints to 0). This should put your arm vertically, where gravity is not a factor. In that configuration, you can apply Zero Torque to all your joints. After that, every torque value should reset to zero. 

Joints can lose calibration with time. We recommand doing this procedure every time before using admittance control, so don't be afraid to it frequently if you need it. 

I can give more precise indications on how to reset torques if you need.

Best,
Felix

Tzxtoiny, 2021-11-25:
> Hi @Tzxtoiny,
>
> You can reset your torques using Development Center (in _Advanced Settings_->_Set Zero Torque_->_apply to all_) or the `in/set_zero_torques` service. This should reset your torques to [...]
Hi，
[image]
@felixmaisonneuve 
Thank you for your reply.
I tried to calibrate the torque sensor. After calibration, I let the manipulator return to the home position, but the output torque is still not zero. The following figure is my output result. I guess the torque sensor calibration is not the reason for the non-zero torque. Is it caused by other forces such as friction? Is there any way to compensate for these forces?

Tzxtoiny, 2021-11-25:
> Hi @Tzxtoiny,
>
> You can reset your torques using Development Center (in _Advanced Settings_->_Set Zero Torque_->_apply to all_) or the `in/set_zero_torques` service. This should reset your torques to [...]

Hi,@felixmaisonneuve
Thank you for your reply.
I tried to calibrate the torque sensor. After calibration, I let the manipulator return to the home position, but the output torque is still not zero. The following figure is my output result. I guess the torque sensor calibration is not the reason for the non-zero torque. Is it caused by other forces such as friction? Is there any way to compensate for these forces?
[image]

Tzxtoiny, 2021-11-25:
> Hi @Tzxtoiny,
>
> You can reset your torques using Development Center (in _Advanced Settings_->_Set Zero Torque_->_apply to all_) or the `in/set_zero_torques` service. This should reset your torques to [...]

Hi,@felixmaisonneuve
Thank you for your reply.
I tried to calibrate the torque sensor. After calibration, I let the manipulator return to the home position, but the output torque is still not zero. The following figure is my output result. 
I guess the torque sensor calibration is not the reason for the non-zero torque. I wonder if the output of `joint_torques_gravity_free` should be approximately zero after the calibration? Or are there still other forces such as friction? Is there any way to compensate for these forces? If so, how to compensate these forces?
[image]

Tzxtoiny, 2021-11-25:
> Hi @Tzxtoiny,
>
> You can reset your torques using Development Center (in _Advanced Settings_->_Set Zero Torque_->_apply to all_) or the `in/set_zero_torques` service. This should reset your torques to [...]

Hi,@felixmaisonneuve
Thank you for your reply.
I tried to calibrate the torque sensor. After calibration, I let the manipulator return to the home position, but the output torque is still not zero. The following figure is my output result.
I guess the torque sensor calibration is not the reason for the non-zero torque. I wonder if the output of joint_torques_gravity_free should be approximately zero after the calibration? Or are there still other forces such as friction? Is there any way to compensate for these forces? If so, how to compensate these forces?
[image]

Tzxtoiny, 2021-11-25:
> Hi @Tzxtoiny,
>
> You can reset your torques using Development Center (in _Advanced Settings_->_Set Zero Torque_->_apply to all_) or the `in/set_zero_torques` service. This should reset your torques to [...]

Thank you for your reply.
I tried to calibrate the torque sensor. After calibration, I let the manipulator return to the home position, but the output torque is still not zero. The following figure is my output result.
I guess the torque sensor calibration is not the reason for the non-zero torque. I wonder if the output of joint_torques_gravity_free should be approximately zero after the calibration? Or are there still other forces such as friction? 
If so, how to compensate these forces?
[image]

felixmaisonneuve, 2021-11-25:
Hi @Tzxtoiny,

This is very weird. For me, all the joints tend to go back to 0 after a movement. Sometimes my third joint will "get stuck" and keep its torque value, but moving the arm a bit in the other direction corrects it and it returns to 0. I am also pretty sure my torque sensor for the third joint has mechanical issues. This is the only joint this happens, so I'm blaming my "broken" sensor, this should not be a problem for you. I don't think there should be other forces to affect torque values.

In a candle-like position (which is 180 degrees for each joint, all torques should be close to 0. This is the values I get 
[image]

When moving to home position after that, I get these values
[image]

I do have to wait a couple seconds before all the values return to 0. 
Did you wait a bit before taking your screenshot? 

Best,
Felix

Tzxtoiny, 2021-11-25:
> Hi @Tzxtoiny,
>
> This is very weird. For me, all the joints tend to go back to 0 after a movement. Sometimes my third joint will "get stuck" and keep its torque value, but moving the arm a bit in the [...]

Hi@felixmaisonneuve
I did wait a long time, but the "joint_torque_gravity_free" still didn't reach zero, especially the torque value after the third joint "gravity_free" was large. 
The following screenshot is the **"gravity_free"** torque I output of 6 joints in home position. All values are approximately 0 except joint 3.
[image]

In your second screenshot, at home, why is the total torque "joint_torques" zero and "joint_gravity_free_torques" not zero? Is it wrong? Isn't it only at the “Candle” position that the total joint torque will be zero?

Tzxtoiny, 2021-11-25:
> Hi @Tzxtoiny,
>
> This is very weird. For me, all the joints tend to go back to 0 after a movement. Sometimes my third joint will "get stuck" and keep its torque value, but moving the arm a bit in the [...]

Hi@felixmaisonneuve
I did wait a long time, but the "joint_torque_gravity_free" still didn't reach zero, especially the torque value after the third joint "gravity_free" was large.
The following screenshot is the "gravity_free" torque I output of 6 joints in home position. All values are approximately 0 except joint 3.
[image]

In your second screenshot, at home, why is the total torque "joint_torques" zero and "joint_gravity_free_torques" not zero? Is it wrong? Isn't it only at the “Candle” position that the total joint torque will be zero?

felixmaisonneuve, 2021-11-26:
Hi @Tzxtoiny,

`joint_torque` should always be 0 unless you apply external forces, In which case it will return torques from said external force.
`joint_torques_gravity_compensation` is the torque values generated by gravity, so whatever forces you apply on your arm, these torque values should always be the same for the same configuration. 
`joint_gravity_free_torques` is the acctual torque for each joint, which includes torque to compensate external forces and gravity.

If you look at my first screenshot, my arm was in home position and I was pushing it (creating more torques on some joints). You can see that torques values from `joint_torques_gravity_compensation` are the same for my last screenshot because, for both screenshots, the arm was in home position (same configuration = same gravity torques). 
Back in my first screenshot, pushing the arm was creating additionnal torque for joint 3 and 4. This is what we see in `joint_torque`. Finally, `joint_gravity_free_torques` gives the actual torque every joint has to maintain tfor the arm to stay in place. 

I read our exchange again and I think there was a misunderstanding. I think everything is working correctly. The values you are getting are the torques your arm needs to compensate for gravity and stay in place. `joint_gravity_free_torques` is not supposed to be 0 in home position (it should be close to 0 in candle-like position thought). 

Regards,
Felix

Tzxtoiny, 2021-11-29:
> Hi @Tzxtoiny,
>
> `joint_torque` should always be 0 unless you apply external forces, In which case it will return torques from said external force. `joint_torques_gravity_compensation` is the torque [...]
Hi,
I checked your code, and still feel confused. 
In your code, the `joint_gravity_free_torques` topic calls the `getGravityCompansatedTorques`  function, and the `getGravityCompansatedTorques` function calls the `GetAngularForceGravityFree` function in the kinova API. You can see from the API document that the `GetAngularForceGravityFree` function returns the torque, without gravity, of each actuator. 
[image]
[image]
[image]

In my opinion, I think `joint_gravity_free_torques` is the joint torque without the gravity, that is to say: if there are not other forces like the friction, when the arm is static and without external force, `joint_gravity_free_torques` is supposed to be zero.

But when the arm is in the home position and other position, I echo `joint_gravity_free_torques` topic, the result is not near zero, the joint 3 is 1.21. And in your screenshot, I found the same result. In this case, I can't get the exact external force applied to the arm. That is my question.
[image]

What's more, I also found that the topic `/out/tool_wrench` can output the Cartesian force, when there is no external forces, the data is also not zero.  It‘s almost the same with I calculated from the force jacobian using the `joint_gravity_free_torques`. 
[image]

Tzxtoiny, 2021-11-29:
Hi,
I checked your code, and still feel confused.
In your code, the joint_gravity_free_torques topic calls the getGravityCompansatedTorques function, and the getGravityCompansatedTorques function calls the GetAngularForceGravityFree function in the kinova API. You can see from the API document that the GetAngularForceGravityFree function returns the torque, without gravity, of each actuator.
[image]
[image]
[image]

In my opinion, I think joint_gravity_free_torques is the joint torque without the gravity, that is to say: if there are not other forces like the friction, when the arm is static and without external force, joint_gravity_free_torques is supposed to be zero.

But when the arm is in the home position and other position, I echo joint_gravity_free_torques topic, the result is not near zero, the joint 3 is 1.21. And in your screenshot, I found the same result. In this case, I can't get the exact external force applied to the arm. That is my question.
[image]
What's more, I also found that the topic /out/tool_wrench can output the Cartesian force, when there is no external forces, the data is also not zero. It‘s almost the same with I calculated from the force jacobian using the joint_gravity_free_torques.
[image]

Sorry to bother you about this question.

felixmaisonneuve, 2022-01-04:
Hi @Tzxtoiny 

Do you still need support on this topic or can we close this issue?
I will close it in a couple of weeks if I don't get any answer from you.

Best,
Felix

Lu-fm, 2022-02-26:
Well, I think you two are in different channels, and I believe there is ambiguity in the name of API and doc.  
`joint_torque` = 0 when there are no external forces, just like the gravity is zero.
`joint_torques_gravity_free`  is the actual torque of the robot, means "joint torques is free with gravity compansation"
`joint_torques_gravity_compansation` is the torque caused by the gravity.

So, @Tzxtoiny thinks the `joint_torques_gravity_free` includes gravity compansation, because the name expanse like "joint torques measured like there is no gravity". That's it, and you can modify the doc as you said above, dear Felix.

felixmaisonneuve, 2022-02-28:
Hi @Lu-fm and @Tzxtoiny,

Unfortunately, I cannot change the documentation provided with the DevelopmentCenter application. 
I understand this definition could be confusing. However, I feel this thread explains quite thoroughly what each function does, so I will close this issue. 

If someone else ask the same question in the future, I will simply link him to this thread.

Regards,
Felix

closed 2022-02-28 by felixmaisonneuve

Betancourt20, 2023-06-26:
Hell @felixmaisonneuve 

I hope you are doing great. Sorry for asking again almost the same questions but It is not clear to me yet.
I have been reading many times this thread but I cannot get my answers.

I want to use the Cartesian force. I am reading the wrench force with
```Cpp
getCartesianForce(KinovaPose &cart_force)
```
but as mentioned before and explained for you, the value of the wrench force is not zero when there is not external force applied to the end-effector. You explained that is due to the gravity. However, as described in the Kinova documentation, the process to acquire these measurements is the following:

1. Torque from each joint is acquired.
2. Gravity torques from the robot is subtracted.
3.  The Cartesian force is estimated from the joint torques.

Even after subtracting  the gravity, the gravity should appear in the Cartesian force? do you know how do they computed this force (the equations they used) ?

Also, in your first comment in this thread you say that 

```
joint_torque = GetAngularForceGravityFree + GetAngularTorqueGravityEstimation
```
and that ``GetAngularTorqueGravityEstimation`` can be called directly by ``getGravityCompensatedTorques ``. What I do not understand is that in fact ``GetAngularTorqueGravityEstimation `` is used by `` getGravityCompensatedTorques`` as we can verify in 
```cpp
void KinovaComm::getGravityCompensatedTorques(KinovaAngles &tqs)
{
    std::lock_guard<std::recursive_mutex> lock(api_mutex_);
    AngularPosition kinova_tqs;
    memset(&kinova_tqs, 0, sizeof(kinova_tqs));  // zero structure

    int result = kinova_api_.getAngularForceGravityFree(kinova_tqs);
    if (result != NO_ERROR_KINOVA)
    {
        throw KinovaCommException("Could not get the joint torques", result);
    }

    tqs = KinovaAngles(kinova_tqs.Actuators);
}
```
Could you please help me a little bit to understand?

I found in kinova_appi.cpp
```cpp
    getAngularTorqueGravityEstimation = (int (*)(float[COMMAND_SIZE]))initCommandLayerFunction("GetAngularTorqueGravityEstimation");
```
but it is not implemented in kinova_comm, but I guess is not that easy to implement. I believe it will have the same form as 
```cpp
    getAngularTorqueCommand = (int (*)(float[COMMAND_SIZE]))initCommandLayerFunction("GetAngularTorqueCommand");
```
and it respective function in kinova_comm.cpp

Summarizing,  what is 
1. `` getAngularForceGravityFree``
2. `` getAngularTorqueGravityEstimation`` 
3.  `` getGravityCompensatedTorques(KinovaAngles &tqs)``
4.   `` getJointTorques(KinovaAngles &tqs) ``
5.  `` getAngularTorqueCommand (float joint_torque[COMMAND_SIZE]) ``

Thank you so much for your help and patience.
Kind regards,

felixmaisonneuve, 2023-06-27:
Hi @Betancourt20,

I read everything again and got confused. I think part of what I said is incorrect, so I decided to do the same experiment again. 

I will detail the results here and answer you questions afterwards.

First, I created my publishers for API calls `getAngularForce` (joint_torques), `getAngularForceGravityFree` (joint_torques_gravity_free) and `getAngularTorqueGravityEstimation` (joint_torques_gravity_compensation)

In a candle position (all joints to 180deg), I got these values
[image]

In home position, I got these values
[image]

`/out/joint_torques` gives torque read by the actuator
`/out/joint_torques_gravity_compensation` gives torque induced by gravity. This is computed by the arm's controller based on the position of each actuators. These values stay constant as long as the arm does not move. 
`/out/joint_torques_gravity_free` This is the torque on each actuator if we removed the gravity induced torques (joint_torques_gravity_compensation). I think all these torques should be close to 0 unless an external force is applied on an actuator. 

I think this might have been what @Tzxtoiny was saying and I somehow confused `joint_torques` and `joint_torques_gravity_free` and swapped them

To clarify your questions
`getGravityCompensatedTorques` is the kinova_comm function thats calls `getAngularForceGravityFree` (the API call), so `getGravityCompensatedTorques == getAngularForceGravityFree`

`getAngularTorqueGravityEstimation` is a computed estimation of torques induced by the gravity on the arm. This value is estimated from the position of the arm and its model. This is an estimation and it is not fully accurate since the arm is not calibrated.

`getJointTorques` is the kinova_comm functions that returns the actual torque reading (coming from the actuator). These values are updated constantly and is coming directly from the torque sensor on each actuator.

`getAngularTorqueCommand` is something else entirely. When in torque control mode, you can send torques to the arm. This is the actual commands that the arm receive.

 
You also mention getCartesianForce (published on `/out/tool_wrench`). This API calls gives the external forces applied on the end-effector. This wrench value is computed using the `getAngularForceGravityFree` torque values. So documentation is right when it says it "takes torque vector, substract gravity vector and computes wrench with new torque vector". This should also be close to 0 (if no external forces are applied), but it is not very accurate since it depends on `getAngularForceGravityFree` that is not accurate either.

I hope this clarifies a bit

If you need anything else, please ask,
Best Regards,
Felix

felixmaisonneuve, 2023-06-28:
Lets start with `getAngularTorqueGravityEstimation`

`getAngularTorqueGravityEstimation` is computed using the theorical dimensions and mass of the arm. I cannot share the algorithm, but I said it is an estimation because it assumes perfect conditions (dimensions, mass, center of mass, no friction...). When I said the arm is not calibrated, I meant we do not have a calibration file with the exact property of the arm that could allow a better gravity estimation, as well as better cartesian precision. The controller always computes everything assuming the arm is perfect, which is never the case.  The Optimization algorithm will certainly help, but I don't know to what extent.

`getAngularForceGravityFree` is basically the `getJointTorques` values minus the `getAngularTorqueGravityEstimation` values. 

About your figures. Tm5 is not so bad, but it is weird that this happens so soon after a zero torque procedure. It is true torque sensors drift easily, but this one is a bit much. 
What worries me however are the values of `getAngularTorqueGravityEstimation`, aren't they a bit big? In a candle position, every torque should be close to 0. The biggest torque I got was 0.001695. Your gravity estimation for joint 4 is almost -0.8.

For the wrench, it should be close to 0 in every position unless you apply forces on the arm. But, since it depends on the  `getAngularForceGravityFree` torques, it won't be very precise. When I did the expirement, my wrench was not accurate at all. I think I had a force of 1N or 2N applied in one axis.

Betancourt20, 2023-06-29:
Regarding home position, I have the following values:
[image]

So, it's here where I do not understand, why the wrench has values different from zero. There is not external force acting on the end effector. You mentioned that it was related to the gravity, but still I do not see why. Do you believe you can give me a hint with the equations please?

Moreover, I believe in order to use the admittance controller from the API, the signals were filtered right? Do you have at what frequency (I guess it was a bandpass filter, first or second order) ? 

Finally, do you believe it is possible to get the $\tau_{fric}$?

Best regards

felixmaisonneuve, 2023-07-05:
For your first comment, it is very possible the "optimal parameters" are intended only for Torque control mode (I do not know how these parameters are computed and how they are used). In any case, it is hard to acheive any precise work with Jaco2 arms, regardless of the parameters.

For your second comment, wrench is not 0 because it is computed with the torques from `getAngularForceGravityFree`. These torques are not very accurate (this is inherent to the Jaco2), so the resulting wrench will be off as well. I do not think this is related to gravity. I was probably saying that because I confused `joint_torques` and `joint_torques_gravity_free` previously and thought wrench was computed using torques that included gravity, but it is not the case. 

I am not sure what equations you want, but I cannot share the equations to convert actuator torques into wrench or the Kinematics equations. If you are looking for something else, I can maybe help you.

The controller inside the base loops every 10ms (100Hz), so communication with the API must be every 10ms. The base communicates with actuators every ~3ms (300Hz-400Hz), the motor control inside the actuator loops every 1ms (1000Hz).
I am not sure what signal you are talking about and what filtered fontion you are refering to. Motor Control?

I am not sure what friction coefficcient you are looking for either.

Betancourt20, 2023-07-17:
Hi @felixmaisonneuve 

Thank you again for taking the time to answer. I really appreciate your help. 

I am aware is quite difficult to achieve precise work with Jaco2 arms. I am trying (with your help) to do as much as I can to get something clean that can help others to work with this kind of manipulators. I am trying to implement filters and observers to obtain better quality of the signals. I believe this could help to improve the wrench and many other things.

For instance, when the robot is at home position, or other positions, ``getJointVelocities``  gives measurements even if the manipulators is not moving (zero position and Torque zero calibration has been done already). The measurements are not even close to zero. However, the joint angular positions is pretty good. I do not understand why the joint velocity is not that good if it is obtained based on the aforementioned signal. Thus, I am trying to implement a Kalman filter and observer to have a better performance.

Therefore, my following question is mostly related to velocity control. I am trying to read the angular command sent to the controller. I am using the following function
```cpp
/**
 * @brief This function get the angular command of all actuators.
 * @param An AngularPosition struct containing the values. Units are degrees. angular_command {AngularInfo, FingersPosition}
 * @return
 *        - @ref NO_ERROR_KINOVA if operation is a success
 *        - @ref ERROR_API_NOT_INITIALIZED if the API has not been initialized. To initialize it, call the InitAPI() function.
 */
void KinovaComm::getAngularCommand(AngularPosition &angular_command)
{
    std::lock_guard<std::recursive_mutex> lock(api_mutex_);
    memset(&angular_command, 0, sizeof(angular_command));
    int result = kinova_api_.getAngularCommand(angular_command);
    if (result != NO_ERROR_KINOVA)
    {
        throw KinovaCommException("Could not get the angular command", result);
    }
}
```
``getAngularCommand`` gives angular commands positions if the control is set to  angular position. If the control mode is set to angular velocity, then it supposes to give angular velocity commands.
 
I send the joint velocities with
```cpp

/**
 * @brief This function controls robot with joint velocity.
 * This function sends trajectory point(ANGULAR_VELOCITY) that will be added in the robotical arm's FIFO. Waits until the arm has stopped moving before releasing control of the API. sendAdvanceTrajectory() is called in api to complete the motion.
 * @param joint_vel joint velocity in degree/second
 */
void KinovaComm::setJointVelocities(const AngularInfo &joint_vel)
{
    std::lock_guard<std::recursive_mutex> lock(api_mutex_);

    if (isStopped())
    {
        std::cout << "The velocities could not be set because the arm is stopped"<< std::endl;
        return;
    }

    TrajectoryPoint kinova_velocity;
    kinova_velocity.InitStruct();

    memset(&kinova_velocity, 0, sizeof(kinova_velocity));  // zero structure

    kinova_velocity.Position.Type = ANGULAR_VELOCITY;
    kinova_velocity.Position.Actuators = joint_vel;

    int result = kinova_api_.sendAdvanceTrajectory(kinova_velocity);
    if (result != NO_ERROR_KINOVA)
    {
        throw KinovaCommException("Could not send advanced joint velocity trajectory", result);
    }
}
```
The problem is that ``getAngularCommand`` is giving all the time angular commands positions, even if I control mode is set to velocity control (I am sure this control mode is working). Could you give me a hint please?

Sincerely,
Julio

felixmaisonneuve, 2023-07-20:
Yes, getAngularCommand always gives the position command of each actuator. 
You can try using the internal `GetCommandVelocity` API call, but it seems to only send the last delta in position and cartesian value is always 0. I am not sure if I would trust that call.