Kinovarobotics/kinova-ros #369: 3 Finger gripper - Incorrect origin of the 1st finger

AndrejOrsula, 2021-11-30:
## Description

Origin of the 1st finger in the 3-finger gripper has an incorrect origin. This issue has already been reported in https://github.com/Kinovarobotics/kinova-ros/issues/217 back in 2019 but it hasn't been resolved.

Below is the xacro line that causes the problem.

https://github.com/Kinovarobotics/kinova-ros/blob/3078ec6f00225e0ec5b80aae800bd3d94a59c369/kinova_description/urdf/kinova_finger_set.xacro#L25

## Version

ROS distribution : `rolling` (Irrelevant)

Branch and commit you are using : I believe this issue has always been present on all branches

## Expected behavior

3 Finger gripper should have the same kinematic configuration as the real robot, where `finger_1` and `finger_2` close towards each other to create a pinching grasp.

[image]

## Actual behavior

The origin of `finger_1` is incorrect and it points more towards `finger_3` (top) than it should, as seen in the image below with the gripper closed.

[image]

---

Even when using the rotation described in the comment above the xacro line (RPY `-1.5707963267948966 0.6492624817418906 1.7819811662862104`), the fingers still do not close properly as they should.

https://github.com/Kinovarobotics/kinova-ros/blob/3078ec6f00225e0ec5b80aae800bd3d94a59c369/kinova_description/urdf/kinova_finger_set.xacro#L24-L25

- **Left image**: Current version (pointing towards incorrect `finger_3`)
- **Right image**: Using `-1.5707963267948966 0.6492624817418906 1.7819811662862104`, `finger_1` points towards the correct `finger_2` but it is still wrong and the fingers do not close as they should.
<p align="center" float="middle">
  [image]
  [image]
</p>

AndrejOrsula, 2021-11-30:
If someone wants to tackle this issue, [Kinova Gen2 Technical Resources](https://www.kinovarobotics.com/en/resources/gen2-technical-resources) contains link to download the CAD model (`STEP`). However, I do not know if the CAD model is correct.

felixmaisonneuve, 2021-12-01:
Hi @AndrejOrsula,

Thank you for this very detailed issue. I understand when you say the first finger is closing towards the third finger, and this is true (and it should be fixed). 

However, I am not sure what you mean when you say that, even with the correct orientation, the finger is still wrong.
> finger_1 points towards the correct finger_2 but it is still wrong and the fingers do not close as they should

What is still wrong? Can you elaborate on that? 
I want to fully understand what you think should be the correct way to address this issue. If it is just a matther of changing fingers orientation, I can change it pretty quickly, but if the fix requires a "big" amount of work (like modifiying CAD files), it might take some time before I actually tackle this.

Best,
Felix

felixmaisonneuve, 2021-12-02:
Hi @AndrejOrsula,

Doing some Trial and error, I came up with this combination : 
finger 1 : `finger_origin_rpy="-1.570796327 .649262481663582 1.855"`
finger 2 : `finger_origin_rpy="-1.570796327 .649262481663582 -1.2"`

and it resulted in this :
[image]

I don't know where the previous values come from. Simply changing the orientation seems to work just fine. 

Try this combination and see if it reflects your real arm gripper.

Best,
Felix

AndrejOrsula, 2021-12-02:
Hello @felixmaisonneuve,

Thank you for your time! :smile: Your configuration is definitely an improvement over the current version, and it provides better resemblance to the real gripper. With the angles you provided, the fingers can now be used to form a pinching grasp.

However, due to the manual trial-and-error approach, the contact surfaces of the finger tips are not exactly parallel as they would be on the real gripper. It is not currently my use-case, but I think that the tilted contact surfaces could cause some issues when trying to simulate manipulation of small objects (object could eventually slip out of the gripper when using physics engines that implement rigid-body dynamics). 

Therefore, I believe that analysing CAD model (STEP) in an appropriate software is the only way to get that "exact" origin of `finger_1`. I unfortunately do not currently have access to a setup for doing that.

---

I will just provide more information about why I think that origin of `finger_2` and `finger_3` are correct and should not be modified, whereas `finger_1` requires more than one RPY angle (not just yaw) to be corrected.

When looking at `finger_1`, it is visible that the finger mesh does not properly align with its joint on the base of the hand. Below is an image demonstrating that. This image provides a view from behind the finger when the gripper is closed, using your configuration. Image on the right highlights the lines of what I mean (the finger is tilted with respect to the joint that it's connected to).

<p align="center" float="middle">
  [image]
  [image]
</p>

It is also visible on the inner side when the hand is open (image from https://github.com/Kinovarobotics/kinova-ros/issues/217 using the original configuration).

<p align="center" float="middle">
  [image]
</p>

However, when looking at `finger_3`, it seems to me that the finger is seated properly (again - illustrated with lines below). Same applies to `finger_2` using the original configuration.

<p align="center" float="middle">
  [image]
  [image]
</p>

AndrejOrsula, 2021-12-02:
I just found this one - [abr/abr_control/**/jaco2.xml#L130](https://github.com/abr/abr_control/blob/58aa61f40a083db705b8b6d51b53da50ae0edac5/abr_control/arms/jaco2/jaco2.xml#L130). Unfortunately, it uses completely different coordinates, so it's not a straightforward conversion. The script below does the trick.

```python
#!/usr/bin/env python3
import numpy as np
from scipy.spatial.transform import Rotation

rot1 = Rotation.from_quat([0.262085, 0.02762996, -0.10213274, 0.95922726])
rot2 = Rotation.from_euler("xyz", [0.0, -0.5 * np.pi, 0.0])

rpy = (rot1 * rot2).as_euler("xyz")

rpy_xacro_coordinates = (rpy[0] - np.pi, -rpy[1], -rpy[2])
```

The `pitch` (joint position offset) is still a bit wrong because they use a slightly different initial position. I adjusted it manually such that `finger_1` and `finger_2` meet when they have the same joint position (`~1.35`).

**`-1.7047873384941834 0.6476144647144773 1.67317415161155`** seems to be around the desired value. Of course it might be a bit off due to manual adjustment of initial joint position and floating-point precision in the descriptor that I found.
[image]

With this change, it seems to be seated properly
[image]

---

The only thing is that finger tips point slightly in the opposite directions (difficult to show - so you need to see for yourself). I am not sure, but I think it is just that the finger tip does is not rounded in the used mesh.

[image]

closed 2021-12-03 by felixmaisonneuve

felixmaisonneuve, 2021-12-03:
Thank you @AndrejOrsula for this solid work. We really appreciate it. 
I tested it and it looks pretty good. 

About the finger tips, I am not sure what you reffer to. 
If you are talking about this line, I think this is simply an optical illusion
[image]

Gripper fingertips aren't square (left), they are more with this "kind of" octogonal shape (right). 
[image]

I think it creates an illusion from certain angles that gives the impression the fingertip is pointing in the opposite direction. 
Please correct me if I am not looking at the right thing. 

Other than that, thank you again for your support. I already merge both of your pull requests.

Regards,
Felix

AndrejOrsula, 2021-12-03:
Yes @felixmaisonneuve. That was what I meant, and I think your explanation is correct. Thank you.

PR #370 (2021-12-03): [melodic] Fix origin of 1st finger in 3 finger gripper
Resolves #369 

Please, double-check before merge.
PR #371 (2021-12-03): [noetic] Fix origin of 1st finger in 3 finger gripper
Resolves #369 

Please, double-check before merge.