Kinovarobotics/kinova-ros #372: How to speed up the sending rate of arm publishers

Yang12308, 2021-12-09:
I used the command rostopic hz <target_topic_name>（like '/j2n6s300_driver/out/joint_state'). And I found that the average rate is 10. Can increase speed of rate to 100. Thx

felixmaisonneuve, 2021-12-09:
Hi @Yang12308,

In the kinova_driver/src/kinova_arm.cpp file, in the constructor (`KinovaArm()`), the driver defines a param `status_interval_seconds` which correspond to the frequency of the output topics. 

You need to change this line : 
https://github.com/Kinovarobotics/kinova-ros/blob/6b7a12199b29dc7c9ee484eb890bd89e9cf20cd9/kinova_driver/src/kinova_arm.cpp#L224

to this 
```
    node_handle_.param<double>("status_interval_seconds", status_interval_seconds_, 0.01);
```

and recompile your driver with `catkin_make`. After that, your output frequency should be 100Hz.

If you don't want to recompile the driver, it might be possible to simply edit the `status_interval_seconds` param after the driver is launched, but I haven't tried it, so I don't know if it will work. After launching the driver, simply run 
```
rosparam set status_interval_seconds 0.01
```

Regards,
Felix

felixmaisonneuve, 2021-12-09:
Also, there is this old issue #281.

To summarize : 
There is param `feedback_publish_rate` in the launch file that you can use to change the rate
```
roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300 feedback_publish_rate:=0.01
```

Like @wxmerkt said, you might not be able to reach 100Hz. You can comment some of functions that communicates with the arm to accelerate the response time. 
Topics are published by those 4 functions : 
https://github.com/Kinovarobotics/kinova-ros/blob/6b7a12199b29dc7c9ee484eb890bd89e9cf20cd9/kinova_driver/src/kinova_arm.cpp#L759-L765

For example, if you only plan to work in angular mode, you could comment `publishToolPosition()`, `publishToolWrench()` and `publishFingerPosition()` (that you won't need). 

This will limit the number of call with `kinova_comm_.something()` and it will increase your refresh rate.

Yang12308, 2021-12-10:
It worked ！First i connected via Ethernet Then i used this to start
`roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300 feedback_publish_rate:=0.01`
and Commented some of functions
    ```
     publishToolPosition(); 
     publishToolWrench(); 
     publishFingerPosition(); 
```
bacause i just want the information of angle and velocity for joints
The output frequency can be 100Hz.

Thanks,all!

closed 2021-12-10 by felixmaisonneuve