Kinovarobotics/kinova-ros #373: What's the torque range of torque command?

yuweiDu, 2021-12-16:
Hi!
I try to use impedance controller to control the robot and the controller will output torque command to robot. However, when I run the code, the robot don't move at all. The torque command I send to the robot is shown in the picture.
[image]
 Besides, in robot_parameters.yaml , the torque_min and torque_max are commented by default. In this condition, what will happen  when I call `SetTorqueControlParameters` with the service in/set_torque_control_parameters. I have tried, but I can't tell any differences.
    ```
    # torque_min: [80, 80, 80, 80, 80, 80, 80],
    # torque_max: [90, 90, 90, 90, 90, 90, 90],
```
What's more,` publish_torque_with_gravity_compensation: false` is in robot_parameters.yaml. When I change it to true, the robot will move extremely fast in position control mode. Why does this happen?

Thank you!

felixmaisonneuve, 2021-12-16:
Hi @yuweiDu,

You are trying to use impedance on the arm. What code do you use? Before sending torque commands, can you move the arm "by hand" using impedance? 

For `SetTorqueControlParameters`, my previous answer still holds : https://github.com/Kinovarobotics/kinova-ros/issues/365#issuecomment-968922025.
If all the torque_parameters in robot_parameters.yaml are commented and you haven't manually set them, `SetTorqueControlParameters` won't do anything. `SetTorqueControlParameters` will use those parameters and change the behaviour according to them. 

Is it a more clear explanation?

Finally, about `publish_torque_with_gravity_compensation`, this is extremely weird. This parameter should not impact the arm's behaviour at all. It should only cange the output of the topic `out/joint_torques`. 
Did you do other modifications?

Best,
Felix

yuweiDu, 2021-12-27:
Hi! @felixmaisonneuve 
I have tried my code several times and I found that the robot didn't moved because it didn't switch to torque mode successfully every time. 

I have tried to calibrate the torque sensor(sending the robot to cadle-like position and zero-torque) before switching to torque mode, but it couldn't sucessful every time. 

I also used the torque console, and it worked well. However, it seems there are some conflict between torque console and ROS.When I change parameters or mode in torque console, some errors will be encountered(see the pictures below). 

[image]
[image]

In conclusion, I have several questions:
1.What's the differences between`rosservice call  /j2n6s300_driver/in/set_torque_control_mode 1`and `rosservice call /j2n6s300_driver/in/start_force_control`
2.How can I make sure that the robot can switch to torque mode successfully in my python code with ROS ?

Thank you very much!

felixmaisonneuve, 2022-01-03:
Hi @yuweiDu,

There is conflict between Torqueonsole and ROS and it is not recommanded to use both at the same time. Same applies for DevelopmentCenter and ROS. DevelopmentCenter and TorqueConsole can be used together, but you cannot use ROS while one of the two are running. 

1. I am not sure what the difference is between the two. They use different API calls, one service calls `switchTrajectoryTorque` and the other calls `startForceControl`. In my experience, I have had more success using `set_torque_control_mode` (that calls `switchTrajectoryTorque`). I wasn't able to make anything work when using `start_force_control`. 

After looking at the controller's code, both functions do very different things using different variables. In `startForceControl`, I found a nice comment `// TO VALIDATE BIG TIME` that might indicate the function is not fully operationnal. However, I am not exactly sure what that functions does/is supposed to do. I would recommend using `set_torque_control_mode`.

2. I would make sure DevelopmentCenter and TorqueConsole are not opened, and I would follow the example `gravity_compensated_mode.py` (in the kinova_demo folder).

This example puts the arm in candle-like position (1), set all torques to 0 (2) and set the arm in torque control mode using `SetTorqueControlMode` (3). Thoses 3 steps are essential if you want a (somewhat) reliable way to put your arm in torque control mode. 

You can look at `robot_control_module.py` (also in the kinova_demo folder) to see how some of functions of gravity_compensated_mode.py are implemented.

Hope this helps,
Felix

closed 2022-06-08 by felixmaisonneuve