Kinovarobotics/kinova-ros #374: CAD model of two-fingered gripper

hustfzt2, 2021-12-19:
Hi, only the 3-fingered gripper can be available in the Kinova-description file, how can i get the cad model of two-fingered gripper?

felixmaisonneuve, 2021-12-20:
Hi @hustfzt2,

Can you please give more details on what you are looking for?

What Kinova-description file are you talking about. If you look in the kinova_description/meshes/ folder, you will find the CAD file hand_2finger.STL for the 2 fingers gripper. Is that what you are looking for?

You can simulate a Jaco2 with a 2 fingers gripper using Gazebo, it looks like this. 
[image]

Best,
Felix

felixmaisonneuve, 2022-04-12:
Hi @hustfzt2,

Do you still need support for this issue or can we close it?

I will leave this issue open for a bit and close it if I don't get any answer.

Best,
Felix

closed 2022-06-08 by felixmaisonneuve