Kinovarobotics/kinova-ros #378: How to save and load a PRM map

s265452, 2022-01-17:
## Description
Hi everyone,
I'm using a Kinova Jaco2 mod. j2s7s300
ROS Melodic 18.04

I wish to save and load a PRMstar map, so I wrote something like this into ompl_planning.yaml:

  PersistentPRMstar: # use this with a representative environment to create a roadmap
    type: geometric::PRMstar
    multi_query_planning_enabled: true
    store_planner_data: true
    load_planner_data: false
    planner_data_path: /tmp/roadmap.graph
    enforce_joint_model_state_space: true
  SemiPersistentPRMstar: # reuses roadmap during lifetime of node but doesn't save/load roadmap to/from disk
    type: geometric::PRMstar
    multi_query_planning_enabled: true
    store_planner_data: false
    load_planner_data: true
    planner_data_path: /tmp/roadmap.graph
    enforce_joint_model_state_space: true

But when I use  SemiPersistentPRMstar planner, it generate a new roadmap everytime.

How can I fix this problem?

felixmaisonneuve, 2022-01-17:
Hi @s265452,

Just to get it out of the way : Is the file correctly indented?

When using PersistentPRMStar, is the planner data correctly created in /tmp/roadmap.graph?
When using SemiPersistentPRMstar, does /tmp/roadmap.graph exist?
When launching the driver, or when using the planners, do you see any error message?

From what I see, it looks good, however I cannot test it on my side at the moment. 

Regards,
Felix

s265452, 2022-01-17:
No. If I go in the tmp folder

/Computer/tmp/

I don't find the file roadmap.graph

When I use SemiPersistentPRMStar I read:
```

[ INFO] [1642442204.349905255]: Planner configuration 'arm[SemiPersistentPRMstar]' will use planner 'geometric::PRMstar'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1642442204.349978478]: arm[SemiPersistentPRMstar]: problem definition is not set, deferring setup completion...
[ INFO] [1642442204.791611341]: arm[SemiPersistentPRMstar]: Starting planning with 2 states already in datastructure
[ INFO] [1642442219.471138480]: arm[SemiPersistentPRMstar]: Created 258 states
[ INFO] [1642442219.471348465]: Solution found in 15.121318 seconds
[ INFO] [1642442219.471444057]: SimpleSetup: Path simplification took 0.000007 seconds and changed from 2 to 2 states
[ INFO] [1642442219.473740616]: Planned to path constraints. Resuming original planning request.
[ INFO] [1642442219.487210332]: Planner configuration 'arm[SemiPersistentPRMstar]' will use planner 'geometric::PRMstar'. Additional configuration parameters will be set when the planner is constructed.
[ INFO] [1642442219.487299335]: arm[SemiPersistentPRMstar]: problem definition is not set, deferring setup completion...
[ INFO] [1642442219.954660160]: arm[SemiPersistentPRMstar]: Starting planning with 2 states already in datastructure
[ INFO] [1642442219.954815302]: arm[SemiPersistentPRMstar]: Allocating specialized state sampler for state space
[ INFO] [1642442219.954898038]: arm[SemiPersistentPRMstar]: Allocating specialized state sampler for state space
[ INFO] [1642442234.521004564]: arm[SemiPersistentPRMstar]: Created 407 states
[ INFO] [1642442234.521067100]: Solution found in 15.033728 seconds
[ INFO] [1642442234.678464672]: SimpleSetup: Path simplification took 0.157322 seconds and changed from 4 to 6 states
[ INFO] [1642442234.678939334]: Planning adapters have added states at index positions: [ 0 1 2 3 4 5 6 7 8 9 10 11 12 13 14 15 ]
```
Regards

felixmaisonneuve, 2022-01-17:
Does the `/tmp/` folder exist?

Also, since it is in the root directory, moveit might not have the permission to write files in there.
I would suggest to create a folder in your user directory (e.g. `/home/<user>/tmp/`) and use this folder instead. 
```
planner_data_path: /home/<user>/tmp/roadmap.graph
```

After that, check if a file is correctly created when you use PersistentPRMstar

s265452, 2022-01-17:
Thank you for your support but It's the same. 
No file are created in the folder

planner_data_path: /home/francesco/tmp/roadmap.graph 

[image]

felixmaisonneuve, 2022-01-17:
Can you share your full ompl_planning.yaml?

s265452, 2022-01-17:
```
planner_configs:
  SBLkConfigDefault:
    type: geometric::SBL
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    enforce_joint_model_state_space: true
  ESTkConfigDefault:
    type: geometric::EST
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0 setup()
    goal_bias: 0.05  # When close to goal select goal, with this probability. default: 0.05
    enforce_joint_model_state_space: true
  LBKPIECEkConfigDefault:
    type: geometric::LBKPIECE
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    border_fraction: 0.9  # Fraction of time focused on boarder default: 0.9
    min_valid_path_fraction: 0.5  # Accept partially valid moves above fraction. default: 0.5
    enforce_joint_model_state_space: true
  BKPIECEkConfigDefault:
    type: geometric::BKPIECE
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    border_fraction: 0.9  # Fraction of time focused on boarder default: 0.9
    failed_expansion_score_factor: 0.5  # When extending motion fails, scale score by factor. default: 0.5
    min_valid_path_fraction: 0.5  # Accept partially valid moves above fraction. default: 0.5
    enforce_joint_model_state_space: true
  KPIECEkConfigDefault:
    type: geometric::KPIECE
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    goal_bias: 0.05  # When close to goal select goal, with this probability. default: 0.05 
    border_fraction: 0.9  # Fraction of time focused on boarder default: 0.9 (0.0,1.]
    failed_expansion_score_factor: 0.5  # When extending motion fails, scale score by factor. default: 0.5
    min_valid_path_fraction: 0.5  # Accept partially valid moves above fraction. default: 0.5
    enforce_joint_model_state_space: true
  RRTkConfigDefault:
    type: geometric::RRT
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    goal_bias: 0.05  # When close to goal select goal, with this probability? default: 0.05
    enforce_joint_model_state_space: true
  RRTConnectkConfigDefault:
    type: geometric::RRTConnect
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    enforce_joint_model_state_space: true
  RRTstarkConfigDefault:
    type: geometric::RRTstar
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    goal_bias: 0.05  # When close to goal select goal, with this probability? default: 0.05
    delay_collision_checking: 1  # Stop collision checking as soon as C-free parent found. default 1
    enforce_joint_model_state_space: true
  TRRTkConfigDefault:
    type: geometric::TRRT
    optimization_objective: MechanicalWorkOptimization  #MaximizeMinClearanceObjective
    range: 0.0  # Max motion added to tree. ==> maxDistance_ default: 0.0, if 0.0, set on setup()
    goal_bias: 0.05  # When close to goal select goal, with this probability? default: 0.05
    max_states_failed: 10  # when to start increasing temp. default: 10
    temp_change_factor: 2.0  # how much to increase or decrease temp. default: 2.0
    min_temperature: 10e-10  # lower limit of temp change. default: 10e-10
    init_temperature: 10e-6  # initial temperature. default: 10e-6
    frountier_threshold: 0.0  # dist new state to nearest neighbor to disqualify as frontier. default: 0.0 set in setup() 
    frountierNodeRatio: 0.1  # 1/10, or 1 nonfrontier for every 10 frontier. default: 0.1
    k_constant: 0.0  # value used to normalize expresssion. default: 0.0 set in setup()
    enforce_joint_model_state_space: true
  PRMkConfigDefault:
    type: geometric::PRM
    max_nearest_neighbors: 10  # use k nearest neighbors. default: 10
    enforce_joint_model_state_space: true
  PRMstarkConfigDefault:
    type: geometric::PRMstar
    enforce_joint_model_state_space: true
  PersistentPRMstar: # use this with a representative environment to create a roadmap
    type: geometric::PRMstar
    multi_query_planning_enabled: true
    store_planner_data: true
    load_planner_data: false
    planner_data_path: /home/francesco/tmp/roadmap.graph
    enforce_joint_model_state_space: true
  SemiPersistentPRMstar: # reuses roadmap during lifetime of node but doesn't save/load roadmap to/from disk
    type: geometric::PRMstar
    multi_query_planning_enabled: true
    store_planner_data: false
    load_planner_data: true
    planner_data_path: /home/francesco/tmp/roadmap.graph
    enforce_joint_model_state_space: true

arm:
  planner_configs:
    - SBLkConfigDefault
    - ESTkConfigDefault
    - LBKPIECEkConfigDefault
    - BKPIECEkConfigDefault
    - KPIECEkConfigDefault
    - RRTkConfigDefault
    - RRTConnectkConfigDefault
    - RRTstarkConfigDefault
    - TRRTkConfigDefault
    - PRMkConfigDefault
    - PRMstarkConfigDefault
    - PersistentPRMstar
    - SemiPersistentPRMstar

gripper:
  planner_configs:
    - SBLkConfigDefault
    - ESTkConfigDefault
    - LBKPIECEkConfigDefault
    - BKPIECEkConfigDefault
    - KPIECEkConfigDefault
    - RRTkConfigDefault
    - RRTConnectkConfigDefault
    - RRTstarkConfigDefault
    - TRRTkConfigDefault
    - PRMkConfigDefault
    - PRMstarkConfigDefault
```

felixmaisonneuve, 2022-01-17:
This file seems pretty sensitive to the format, I would try to put PersistentPRMstar at the top (before SBLkConfigDefault) and try again to see if it creates a graph file.

I would also try different combinaisons in the parameters order (e.g. type->enforce_joint_model_state_space->multi_query_planning_enabled->planner_data_path->store_planner_data->load_planner_data).

I cannot test right now and I find it very frustrating. If the issue is not resolved by then, I will try to test on an arm later this week,

s265452, 2022-01-18:
I did your modification but the problem still persist.

Is it possible that when I launch` j2s7s300_virtual_robot_demo.launch` I have to specify `db:=true`?

Even if I launch:

`roslaunch j2s7s300_moveit_config j2s7s300_virtual_robot_demo.launch robotType:=j2s7s300 db:=true`

I have this error

```

 * /warehouse_exec: mongod
 * /warehouse_host: localhost
 * /warehouse_plugin: warehouse_ros_mon...
 * /warehouse_port: 33829

NODES
  /
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    mongo_wrapper_ros_francesco_N552VX_27568_7264577604552829809 (warehouse_ros/mongo_wrapper_ros.py)
    move_group (moveit_ros_move_group/move_group)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz_francesco_N552VX_27568_8437795419075632837 (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[joint_state_publisher-1]: started with pid [27620]
process[robot_state_publisher-2]: started with pid [27621]
process[move_group-3]: started with pid [27622]
process[rviz_francesco_N552VX_27568_8437795419075632837-4]: started with pid [27623]
ERROR: cannot launch node of type [warehouse_ros/mongo_wrapper_ros.py]: Cannot locate node of type [mongo_wrapper_ros.py] in package [warehouse_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
[ERROR] [1642504676.816938825]: Could not find the 'robot' element in the xml file
[ERROR] [1642504676.862846346]: Could not find the 'robot' element in the xml file
[ERROR] [1642504676.863845999]: Unable to parse URDF from parameter '/robot_description'
[ERROR] [1642504676.915381956]: Robot model not loaded
[ERROR] [1642504676.927382450]: Planning scene not configured
[ INFO] [1642504676.968919278]: rviz version 1.13.23
[ INFO] [1642504676.968975459]: compiled against Qt version 5.9.5
[ INFO] [1642504676.968988087]: compiled against OGRE version 1.9.0 (Ghadamon)
```

felixmaisonneuve, 2022-01-18:
This is very possible.

Do you have the correct packages installed?
`sudo apt-get install ros-noetic-warehouse-ros-mongo`

After, according MoveIt documentation on [Warehouse](https://ros-planning.github.io/moveit_tutorials/doc/persistent_scenes_and_states/persistent_scenes_and_states.html), you will need to add this line in kinova_moveit/robot_configs/j2s7s300_moveit_config/launch/warehouse_settings.launch.xml
```
  <param name="warehouse_plugin" value="warehouse_ros_mongo::MongoDatabaseConnection"
```

This line already exists in the j2s6s300 config, but not on the other robot configs
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_moveit/robot_configs/j2s6s300_moveit_config/launch/warehouse_settings.launch.xml#L14

If it works, I will update the repo to add this line in all of our configs.

Start with this and let me know how it goes

Best,
Felix

s265452, 2022-01-18:
I did these modifications.

This is my 

kinova_moveit/robot_configs/j2s7s300_moveit_config/launch/warehouse_settings.launch.xml
```

<launch>
  <!-- Set the parameters for the warehouse and run the mongodb server. -->

  <!-- The default DB port for moveit (not default MongoDB port to avoid potential conflicts) -->  
  <arg name="moveit_warehouse_port" default="33829" /> 

  <!-- The default DB host for moveit -->
  <arg name="moveit_warehouse_host" default="localhost" /> 
  
  <!-- Set parameters for the warehouse -->
  <param name="warehouse_port" value="$(arg moveit_warehouse_port)"/>
  <param name="warehouse_host" value="$(arg moveit_warehouse_host)"/>
  <param name="warehouse_exec" value="mongod" />
  <param name="robot_description" command="$(find xacro)/xacro --inorder '$(find kinova_description)/urdf/j2s7s300.xacro'" />
  <param name="warehouse_plugin" value="warehouse_ros_mongo::MongoDatabaseConnection" />

</launch>
```

felixmaisonneuve, 2022-01-20:
Hi @s265452,

I did not forget you. I am doing some testing and found a couple of things : 
1. In warehouse.launch, you need to change `warehouse_ros` to `warehouse_ros_mongo`
```
  <node name="$(anon mongo_wrapper_ros)" cwd="ROS_HOME" type="mongo_wrapper_ros.py" pkg="warehouse_ros_mongo">
```
2. You need to add a dependency in your moveit config package.xml file. You will need to recompile with `catkin_make` after that.
```
  <run_depend>warehouse_ros_mongo</run_depend> 
``` 
I have found this interesting PR (and the related issue) about these changes in moveit : https://github.com/ros-planning/moveit/pull/81/files
3. You need to install mongodb `sudo apt install mongodb-server-core`.
Otherwise you will get errors that "mongod can't be found" 

After all this, I can connect to the database (in the "Context" tab of the "Motion Planning" view in MoveIt) and save/load Scenes and States.

Try this and let me know how it goes.

Best,
Felix

s265452, 2022-01-21:
Hi @felixmaisonneuve ,
I followed your tips but the problem still exists.

It seems that there are ''problems with the 'robot' element in the xml file''

```
[robot_state_publisher-3] restarting process
process[robot_state_publisher-3]: started with pid [10656]
[ERROR] [1642757014.714470276]: Could not find the 'robot' element in the xml file
[robot_state_publisher-3] process has died [pid 10656, exit code 1, cmd /opt/ros/melodic/lib/robot_state_publisher/robot_state_publisher __name:=robot_state_publisher __log:=/home/francesco/.ros/log/c230dec0-7a9b-11ec-b44e-a402b9824c4a/robot_state_publisher-3.log].
log file: /home/francesco/.ros/log/c230dec0-7a9b-11ec-b44e-a402b9824c4a/robot_state_publisher-3*.log
```
Regards.
Francesco

felixmaisonneuve, 2022-01-21:
This is weird, did you do any other moifications in the launch files except the ones I suggested?
Can you copy your full output for 
```
roslaunch j2s7s300_moveit_config j2s7s300_virtual_robot_demo.launch robotType:=j2s7s300 db:=true
```

Param `/robot_description` should be populated by the planing_context.launch, which is called directly in j2s7s300_virtual_robot_demo.launch
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_moveit/robot_configs/j2s7s300_moveit_config/launch/j2s7s300_virtual_robot_demo.launch#L16-L18

This seems to me to be unrelated to the argument `db:=true`. Can you make sure `roslaunch j2s7s300_moveit_config j2s7s300_virtual_robot_demo.launch robotType:=j2s7s300` is running properly?

s265452, 2022-01-21:
No, I didn't other modifications.

When I launch 

`roslaunch j2s7s300_moveit_config j2s7s300_virtual_robot_demo.launch robotType:=j2s7s300  db:=true`

I have these messages

```
 roslaunch j2s7s300_moveit_config j2s7s300_virtual_robot_demo.launch robotType:=j2s7s300  db:=true
... logging to /home/francesco/.ros/log/fd12d15a-7ac9-11ec-9545-a402b9824c4a/roslaunch-francesco-N552VX-4005.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/francesco/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

xacro.py is deprecated; please use xacro instead
xacro: in-order processing became default in ROS Melodic. You can drop the option.
xacro: in-order processing became default in ROS Melodic. You can drop the option.
started roslaunch server http://francesco-N552VX:37619/

SUMMARY
========

PARAMETERS
 * /joint_state_publisher/use_gui: False
 * /joint_state_publisher/zeros/j2s7s300_joint_2: 3.1415
 * /joint_state_publisher/zeros/j2s7s300_joint_4: 3.1415
 * /joint_state_publisher/zeros/j2s7s300_joint_6: 3.1415
 * /mongo_wrapper_ros_francesco_N552VX_4005_826400591204175550/database_path: /home/francesco/c...
 * /mongo_wrapper_ros_francesco_N552VX_4005_826400591204175550/overwrite: False
 * /mongo_wrapper_ros_francesco_N552VX_4005_826400591204175550/robot_description: <?xml version="1....
 * /move_group/allow_trajectory_execution: True
 * /move_group/allowed_execution_duration_scaling: 1.2
 * /move_group/allowed_goal_duration_margin: 0.5
 * /move_group/arm/planner_configs: ['PersistentPRMst...
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'joints': ['j2s...
 * /move_group/gripper/planner_configs: ['SBLkConfigDefau...
 * /move_group/initial: [{'pose': 'Home',...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_fake_contr...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/enforce_joint_model_state_space: True
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/ESTkConfigDefault/enforce_joint_model_state_space: True
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/KPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/KPIECEkConfigDefault/enforce_joint_model_state_space: True
 * /move_group/planner_configs/KPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/KPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/KPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/KPIECEkConfigDefault/type: geometric::KPIECE
 * /move_group/planner_configs/LBKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/LBKPIECEkConfigDefault/enforce_joint_model_state_space: True
 * /move_group/planner_configs/LBKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/LBKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/LBKPIECEkConfigDefault/type: geometric::LBKPIECE
 * /move_group/planner_configs/PRMkConfigDefault/enforce_joint_model_state_space: True
 * /move_group/planner_configs/PRMkConfigDefault/max_nearest_neighbors: 10
[... 509 lines cut ...]
You can start planning now!

[ INFO] [1642776974.511473002]: Loading robot model 'j2s7s300'...
[ WARN] [1642776974.511538407]: Skipping virtual joint 'world_to_base_virtual_joint' because its child frame 'root' does not match the URDF frame 'world'
[ INFO] [1642776974.511570741]: No root/virtual joint specified in SRDF. Assuming fixed joint
[ WARN] [1642776974.774053544]: Kinematics solver doesn't support #attempts anymore, but only a timeout.
Please remove the parameter '/rviz_francesco_N552VX_4398_8102224281572392217/arm/kinematics_solver_attempts' from your configuration.
[ WARN] [1642776974.783459064]: IK plugin for group 'arm' relies on deprecated API. Please implement initialize(RobotModel, ...).
[ INFO] [1642776974.788435608]: IK Using joint j2s7s300_link_1 -3.40282e+38 3.40282e+38
[ INFO] [1642776974.788474424]: IK Using joint j2s7s300_link_2 0.820305 5.46288
[ INFO] [1642776974.788488415]: IK Using joint j2s7s300_link_3 -3.40282e+38 3.40282e+38
[ INFO] [1642776974.788510346]: IK Using joint j2s7s300_link_4 0.523599 5.75959
[ INFO] [1642776974.788524544]: IK Using joint j2s7s300_link_5 -3.40282e+38 3.40282e+38
[ INFO] [1642776974.788541992]: IK Using joint j2s7s300_link_6 1.13446 5.14872
[ INFO] [1642776974.788566175]: IK Using joint j2s7s300_link_7 -3.40282e+38 3.40282e+38
[ INFO] [1642776974.788595361]: Looking in common namespaces for param name: arm/position_only_ik
[ INFO] [1642776974.790096571]: Looking in common namespaces for param name: arm/solve_type
[ INFO] [1642776974.792102611]: Using solve type Manipulation2
[ INFO] [1642776975.257843721]: Starting planning scene monitor
[ INFO] [1642776975.260510090]: Listening to '/move_group/monitored_planning_scene'
[ INFO] [1642776975.283378846]: Constructing new MoveGroup connection for group 'arm' in namespace ''
[ INFO] [1642776976.354168098]: Ready to take commands for planning group arm.
[ INFO] [1642776976.354239693]: Looking around: no
[ INFO] [1642776976.354272025]: Replanning: no
```

felixmaisonneuve, 2022-01-24:
Hi @s265452,

I went and tried it on Ubuntu 18 with ROS Melodic. 
I did not get any errors. I was able to launch moveit with mogodb and connect my planner to it. 

Can you make sure you are using the latest melodic-devel branch and you did not change any files except the one from your specific moveit config folder?
You cannot use the bugfix/moveit-with-mongodb branch I created since it is based on the noetic-devel branch (and it won't work on ros melodic)
Also, I would do a `catkin_make` to make sure everything is correctly compiled (even if this step is not required if you only change launch files).

Point is, I don't know why it is not working for you. I tried the exact steps I suggested on the same setup and everything is working fine. 

To me, the errors you get make no sense with the modifications I suggested, so there must be something else.

Best,
Felix

s265452, 2022-01-24:
HI @felixmaisonneuve,
It seems that the problem was about in the warehouse_settings.launch.xml in the line code:

`<param name="robot_description" command="$(find xacro)/xacro --inorder '$(find kinova_description)/urdf/j2s7s300.xacro'" />`

Now, I'm able to launch:

`roslaunch j2s7s300_moveit_config j2s7s300_virtual_robot_demo.launch robotType:=j2s7s300 db:=true`

I tried to save the roadmap moving the folder in the catkin_ws:

```
planner_configs:
  PersistentPRMstar: # use this with a representative environment to create a roadmap
    type: geometric::PRMstar
    multi_query_planning_enabled: true
    store_planner_data: true
    load_planner_data: false
    planner_data_path: /home/francesco/catkin_ws/tmp/roadmap.graph
  PersistentPRM: # use this to load a previously created roadmap
    type: geometric::PRM
    multi_query_planning_enabled: true
    store_planner_data: false
    load_planner_data: true
    planner_data_path: /home/francesco/catkin_ws/tmp/roadmap.graph
```

It doesn't save the roadmap. 

You are able to save a roadmap.graph ?

felixmaisonneuve, 2022-01-25:
Yes.

If I choose the PersistentPRMstar and move the arm in moveit, once I close the ros driver (CTRL+c in the terminal where you lauched `roslaunch j2s7s300_moveit_config...`), the file `roadmap.graph` will be created.
You do not need to be connected to the database for this to work.

s265452, 2022-01-25:
I followed your instructions but I don't know why, it doesn't work.

felixmaisonneuve, 2022-01-25:
What version of OMPL planner are you using?
According to OMPL planner [documentation](https://ros-planning.github.io/moveit_tutorials/doc/ompl_interface/ompl_interface_tutorial.html?highlight=ompl), you need at least version 1.5 for this to work.
> Note that saving and loading roadmaps is only available in OMPL 1.5.0 and newer. 

You can find your version using `apt-cache policy ros-melodic-ompl`

I am using ros noetic with ompl version 1.5.2 (the most recent one)

s265452, 2022-01-25:
Well, you found the problem.

My version is `1.4.2-5bionic.20191212.045602`

How can I install the new version?

felixmaisonneuve, 2022-01-25:
I don't know if the version 1.5 of the package is available on ros melodic.

maybe `sudo apt-egt update && sudo apt-get upgrade` or `sudo apt-get --reinstall install ros-melodic-ompl`? 
If 1.4.2 is the most recent version for ros melodic on ubuntu 18, I don't know what you could do. I don't see anywhere on the moveit doc that it is, but I don't see anywhere that it isn't either.

radhen, 2024-05-12:
Hello! I have this version of ompl, but I still not able to see any roadmap.graph file. 

> ros-humble-ompl:
  Installed: (none)
  Candidate: 1.6.0-1jammy.20240217.023815
  Version table:
     1.6.0-1jammy.20240217.023815 500
        500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages

I am launching moveit, and selecting the PersistentLazyPRMstar planner. I then set the goal position of the arm, and hit Plan&Execute. The planner is able to find a plan and moves the arm, but I do not see the roadmap.graph file anywhere. I have tried using the default path (/tmp) to store the roadmap, I have also used my /home directory as the save location, but I still do not see the roadmap.graph file anywhere. Could someone please help me with this.

radhen, 2024-05-12:
hello @felixmaisonneuve 

I have **ompl version 1.5.2**. For some reason I am not able to save a roadmap.graph file. I chose the **PRMstarLoadStore** and the **PersistentLazyPRMstar** planner in RViz in Motion Planning window, and also tried changing the path to where the roadmap.graph gets saved, from `/tmp` to `/home/<USER>/tmp`. Could you lease help me with this?