Kinovarobotics/kinova-ros #380: Max acceleration limit

s265452, 2022-02-11:
Hi everyone,
is it possibile to know acceleration limits for each Joint?

I also read the technical guide about the actuators at the links below:

https://drive.google.com/file/d/1zGBHD-AuB9Hli0ppUz9s4Ivs5qCP3QUd/view
https://drive.google.com/file/d/1zIWP6OSXGFw63DdBFXqRiZNRYVB08PcC/view

I didn't find this information.

Waiting for your answer.

felixmaisonneuve, 2022-02-11:
Hi @s265452,

This is not really documented and I had to go look in the code to find something.

According to what I found : 
The base's firmware has an acceleration limit of 200 (degrees/s^2 I assume)
The actuator's firmware has an acceleration limit of 300 (but I am not sure if this value is actually used anywhere).

In either case, each command will have to go through the base first, so I suppose the limit will always be 200.

Jaco's code is getting quite old and it is hard to say with certainty that it is the correct value, but it is the best answer I can give you.

Hope this helps,
Felix

closed 2022-04-12 by felixmaisonneuve

felixmaisonneuve, 2022-04-19:
After looking a bit more about the acceleration limits for the Jaco2 arm I found an acceleration max of 0.64 m^2/s in translation and 2.75 rad^2/s in orientation for cartesian movements