Kinovarobotics/kinova-ros #382: Differences between set and feedback signal

s265452, 2022-02-22:
Hi everyone,
I'm using kinova Jaco 2 7Dof, ROS Melodic.

I wish to plot the joint states differences beetwen set and feedback.
If I set an orientation contraint to the end effector and I don't specify `enforce_joint_model_state_space:true` in ompl_planning.yaml
I lose the control of the manipulator as you can see in the video:

[image]

I wish to see the differences between set and feedback and for these reasons I tried to plot data from the following topics:
/j2s7s300_driver/out/joint_state
/j2s7s300_driver/trajectory_controller/command

It seems that set and feedback have the same behaviour. How Is ti possible?!

felixmaisonneuve, 2022-02-24:
Hi @s265452,

Here is the definition of the `enforce_joint_model_state_space` parameter from the MoveIt [documentation](https://ros-planning.github.io/moveit_tutorials/doc/ompl_interface/ompl_interface_tutorial.html#enforce-planning-in-joint-space)
>Depending on the planning problem MoveIt chooses between joint space and cartesian space for problem representation. Setting the group parameter enforce_joint_model_state_space enforces the use of joint space for all plans.
By default planning requests with orientation path constraints are sampled in cartesian space so that invoking IK serves as a generative sampler.
By enforcing joint space the planning process will use rejection sampling to find valid requests. Please not that this might increase planning time considerably.

I suppose there is a bug in the Trac-IK solver and it doesn't handle correctly your joint constraint to the end effector when solving in cartesian space. My guess is your gripper reaches an angle outside of your constraint, so the tracking finds another trajectory and your arm "flips around" and get stuck.
Forcing to use the joint space with `enforce_joint_model_state_space:true` when solving the IK prevent this.

You could try to monitor only your 7th joint (the one with the constraint) and check if it gets close/outside of your limits.

I feel like this is a bug with Trac-IK. I don't know what I can do to help you. Have you tried using a different IK solver and see if this also happens? You could try to remove/change to constraint as well.

Regards,
Felix

s265452, 2022-02-24:
Mmm.. Thank you @felixmaisonneuve .

I have another question. In order to compute the trajectory, what kind of signal are sent to the manipualtor in order to execute the trajectory? Position? Velocity? Effort?

felixmaisonneuve, 2022-02-24:
I am not sure I understand your question. What do you call the "manipulator"?

MoveIt will publish [trajectory_msgs/JointTrajectory](http://docs.ros.org/en/melodic/api/trajectory_msgs/html/msg/JointTrajectory.html) messages on the `trajectory_controller/command` topic. `JointTrajectory` contains a list of [JointTrajectoryPoint](http://docs.ros.org/en/melodic/api/trajectory_msgs/html/msg/JointTrajectoryPoint.html) that each contains `positions`, `velocities`, `accelerations`, and `efforts`.

Our ros driver will then read the message on the topic and send commands based on the velocity values. You can see the JointTrajectoryController will subscribe to the `trajectory_controller/command` topic and call `CommandCB` when it reads a message.
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_joint_trajectory_controller.cpp#L24

`CommandCB` will then send velocity commands to the arm based on the velocity values received from MoveIt : 
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_joint_trajectory_controller.cpp#L157-L173

Does this answer your question?

Best,
Felix

felixmaisonneuve, 2022-02-24:
Joints are servomotors. 

When using the API, every command is interpreted by the base of the arm (the controller). Then, commands are sent to each actuators. Actuators are servomotors and can handle velocity commands. PIDs of the joints can be changed via the API, but it is not recommanded as it can damage your arm and void your garanty and we do not provide tools to do so.
I will link this other [issue](https://github.com/Kinovarobotics/kinova-ros/issues/381) and this more detailed [answer](https://github.com/Kinovarobotics/kinova-ros/issues/202#issuecomment-857768103) I gave a while ago.

What you are refering in the kinova_control folder is only used by the Gazebo Simulation. Everything you see in there does not represent the actual PIDs of the joints. They are values set to mimic (roughly) the actual movements of the arm when doing a simluation. They are set to avoid strange behaviour like jitterand such. 
Since you are using a real arm, you shouldn't have to bother with this.

I am not sure what `i_clamp` is supposed to represent. I assume this is something Gazebo uses for its simulation.

Does that answer everything?
Felix

s265452, 2022-02-25:
Thank you, but I have another question.

I saw differences between two topics:

/j2s7s300_driver/trajectory_controller/command
/j2s7s300_driver/in/joint_velocity

Can you explain me the reason why I have differences in signal Velocity?

felixmaisonneuve, 2022-02-25:
I am not sure what I am looking at and what your question is. 
You can use both `trajectory_controller/command` and `in/joint_velocity` and the result should be the same. 
When you publish on `trajectory_controller/command`, it eventually publishes on `in/joint_velocity` : 
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_joint_trajectory_controller.cpp#L220
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_joint_trajectory_controller.cpp#L27

s265452, 2022-02-25:
I have this behavior only if I don't specify `enforce_joint_model_state_space:true` in the ompl_planning.yaml.

It is very strange. Can you try to do that?

felixmaisonneuve, 2022-02-25:
What behaviour?

s265452, 2022-02-25:
https://github.com/Kinovarobotics/kinova-ros/issues/382#issuecomment-1050746142

felixmaisonneuve, 2022-02-25:
I do not understand what these graphes are showing. 
What is hapenning? What are you doing? What did you expect would happen?

What seems to be the problem with them?

What is the relation with `enforce_joint_model_state_space:true`? 

Where does these signals come from? Do you send two signals `trajectory_controller/command` and `in/joint_velocity` at the same time/

I simply do not understand what I am looking.

s265452, 2022-02-25:
I'm trying to understand the behaviour of the arm when I plan a movement with orientation constrain without specify `enforce_joint_model_state_space:true ` in the ompl_planning.yaml.

So, I plotted the velocity set signals from 2 topics that I expect should be the same.
I did expect that:

/j2s7s300_driver/trajectory_controller/command
/j2s7s300_driver/in/joint_velocity

should be the same.

How can you see in the image attached before, it isn't. Can you tell me why?

felixmaisonneuve, 2022-02-25:
Ok, I understand now.
Let me explain.

MoveIt is planning trajectories and will send a list of points in topic `trajectory_controller/command`. This is why your graph for this topic looks weird. You should only look at the actual points from the command (in red) :
[image]

 kinova_joint_trajectory_controller.cpp will publish (to `in/joint_velocity`) the velocity for the next point (from the moveit command) until it reaches the `time_from_start` (from the moveit command). It will then move on to the next velocity in the trajectory command an publish it until the next point and so on.
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_joint_trajectory_controller.cpp#L222-L225

Let me illustrate : 
When your trajectory reaches/exceeds the timestamp from the blue point, kinova_joint_trajectory_controller.cpp increments the current trajectory point index (`traj_command_points_index_`) from the trajectory generated by MoveIt and starts publishing the velocity of the next trajectory point (in red). 
It will continue to publish this velocity (from the red point) until it reaches the red timestamp. Then, it will move on to the next trajectory point (in green) and publish this new velocity.
At each peak of the `trajectory_controller/command` curve, the published velocity (`in/joint_velocity`) will change for the velocity of the next peak
[image]

What you are seeing is the normal behaviour and it has nothing to do with `enforce_joint_model_state_space:true`.
What is being published on `trajectory_controller/command` and `in/joint_velocity` is correct. It is the actual command (computed by moveit) that is incorrect when using `enforce_joint_model_state_space:true`. I suspect there is a bug in the Trac-IK solver when putting a joint constaint and, because of that bug, the actual trajectory is wrong and the arm goes nuts.

Don't hesitate if this is still unclear.
Best,
Felix

s265452, 2022-02-25:
Thank you for your explanation and I apologize if I wasn't clear in past messages.

It's very strange that it isn't connected to `enforce_joint_model_state_space:true`.

If I specify that command and I send a planning request with orientation constrain, I see a normal behavior of velocity as you can see in the following image:

[image]

s265452, 2022-02-25:
Ok @felixmaisonneuve thank you very much.

closed 2022-02-25 by s265452