Kinovarobotics/kinova-ros #383: Different position between set and feedback in a path with orientation constrain

s265452, 2022-03-09:
Hi everyone ,
I'm using Kinova Jaco2 7 dof,
ROS Distro Melodic

When I send a planning request with orientation constrain to `move_group` node I see differences between the planned path and the real position of the end-effector.

During the execution of the trajectory with path constrain, if I record the postion signal from these topics:
```
/j2s7s300_driver/out/joint_state
/j2s7s300_driver/trajectory_controller/command
```
I see a behaviour like this:

Thin Lines --> Feedback
Thick Lines --> Set

[image]

For this reason the final position of the rnd-effector is no correct.

Can you explain me the reason why I have this behaviour?  Can I solve this problem in order to obtain a better performance?

Regards

felixmaisonneuve, 2022-03-09:
Hi @s265452,

How big is the difference between the planned path and the real position of the end effector? 
The Jaco arm is not very precise and you can sometimes have a ~2cm difference between the command and the actual position. 

Based on your diagram, it doesn't seem to be a big difference, so I suspect this might only be the actual arm that is not very accurate.

There isn't much you can do about this if it is the issue here. 

Best,
Felix

felixmaisonneuve, 2022-04-12:
Hi @s265452 

Do you think this difference is caused by the arm's imprecision? Can we close this issue?

I will close this issue in a couple of weeks if I don't get an answer from you by then.

Best,
Felix

closed 2022-06-08 by felixmaisonneuve