Kinovarobotics/kinova-ros #388: How to grasp an object with  j2s6s300 in Gazebo?

miyukin73, 2022-05-03:
Hi!
I want to grasp an object (like coke can) with j2s6s300 in Gazebo, and i tried as follows:

roslaunch kinova_gazebo robot_launch.launch kinova_robotType:=j2n6s300

However, 3 fingers keep moving in Gazebo, fingers tremble when they make contact with an object, so cannot grasp it.
(Other robot, like PR2, can grasp with the same program.)
What should I do?

I am a beginner and quite confused about how to use it.
Could you please give me advice. 
Thank you!

https://user-images.githubusercontent.com/63595993/166411360-f8f4f8ba-ace2-4c41-a2d6-85d1875bb808.mov

felixmaisonneuve, 2022-05-04:
Hi @miyukin73,

Unfortunately, it is not possible to grasp object in Gazebo for our Jaco2 arm. Grasping in Gazebo requires code to "stick" items to the gripper when they collide/are close for a small amount of time.

For our Gen3 arms, we use a third party pluggin that we added to our ROS stack that makes it possible to grasp items using our robotiq gripper. We do not have a similar package in our Jaco2 ROS stack. I suppose your PR2 Gazebo simulation contains such package. 

I can link link you to the pluggin we use : https://github.com/Kinovarobotics/ros_kortex/tree/noetic-devel/third_party/gazebo-pkgs
If I am not mistaken, this pluggin is pretty generic and is not specific to robotiq grippers. However, you will probably need to do some modifications if you want to make it work for the Jaco2 gripper. 

Best,
Felix

miyukin73, 2022-05-09:
Dear, @felixmaisonneuve 

Thank you very much for the reply.

I'll try the pluggin you taught me.
Also,  if you know the better simulation for grasping with Jaco2 (like Mujoco?), I'm glad to hear it.

Thank you.

felixmaisonneuve, 2022-05-09:
Unfortunately, I do not know any other pluggin or repository that could help you grasp object with the Jaco2 arm in a Gazebo simulation.

Best, 
Felix

felixmaisonneuve, 2022-06-08:
Hi @miyukin73,

Do you still need support about this issue? Did you manage to make the grasping work?
I will probably close it in a couple of weeks if I do not hear from you.

Best,
Felix

closed 2022-08-15 by felixmaisonneuve

xmaww, 2023-12-27:
Hello @miyukin73  @felixmaisonneuve  I would like to ask how you control the movement of the robotic arm before the hand closes in the simulation? Specifically, how to control the robotic arm to reach the right position to grab the Coke can? Is there a position control API function that is directly available?
Many thanks if you can reply!