Kinovarobotics/kinova-ros #389: Clean noetic migration

JorgeEdGG, 2022-05-16:
Working on a clean Ubuntu 20 installation i encounter some little errors related to the python version using gazebo demo. Since ROS-noetic packages are installed over python3 we require the package python-is-python3

Python scrip at src/kinova-ros/kinova_control/src/move_robot.py at line 86 uses python2 print function, it require parentesis as is in python3.

Planning contex launch, at 
src/kinova-ros//kinova_moveit/robot_configs/j2n6s300_moveit/launch/planning_contex.launch
src/kinova-ros//kinova_moveit/robot_configs/j2s6s300_moveit/launch/planning_contex.launch
src/kinova-ros//kinova_moveit/robot_configs/j2s7s300_moveit/launch/planning_contex.launch
src/kinova-ros//kinova_moveit/robot_configs/m1n6s300_moveit/launch/planning_contex.launch

at line 8 require change xacro.py by xacro since xacro.py is deprecated

With this liitle changes gazebo demo will work property.

felixmaisonneuve, 2022-05-16:
Hi @JorgeEdGG,

I think you are using our `melodic-devel` branch.
[move_robot.py](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_control/src/move_robot.py#L86) on melodic-devel vs [move_robot.py](https://github.com/Kinovarobotics/kinova-ros/blob/noetic-devel/kinova_control/src/move_robot.py#L86) on noetic-devel
[planning_contex.launch](https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_moveit/robot_configs/j2n6s300_moveit_config/launch/planning_context.launch#L9) on melodic-devel vs [planning_contex.launch](https://github.com/Kinovarobotics/kinova-ros/blob/noetic-devel/kinova_moveit/robot_configs/j2n6s300_moveit_config/launch/planning_context.launch#L9) on noetic-devel

What you see was already fixed when when we added ROS Noetic support.
Make sure you are using the correct branch and the most recent commit. There was quite a lot of files (not just the ones you mentionned) that needed some changes regarding the Python version and you might experience some other issues.

Best,
Felix

felixmaisonneuve, 2022-06-08:
Hi @JorgeEdGG,

I assume you were using the melodic-devel branch (the default one) instead of our noetic-devel branch. 
I will close this issue, but feel free to re-open it or leave a comment if you have any question.

Best,
Felix

closed 2022-06-08 by felixmaisonneuve