Kinovarobotics/kinova-ros #390: Flashing green light after firmware update.

ericdusel77, 2022-05-23:
I have a flashing green light that does not go away and I cannot use the arm.

I have tried updating the firmware multiple times and it says it was successful.  It says the connection is good in the Torque Console.

When I launch the kinova_bringup roslaunch file, I get a line that says:
`Found 1 device(s), using device at index 0 (model:  ????oU, serial number:@?YSK??, code version: 0, code revision: 0`

ericdusel77, 2022-05-23:
Just looked at some old logs, that line USED to say...

`Found 1 device(s), using device at index 0 (model: JACO REHAB V1, serial number: Not Set, code version: 327936, code revision: 2)`

felixmaisonneuve, 2022-05-24:
Hi @ericdusel77,

I have this kind of error when playing with Development Center and ROS at the same time. I think I also got something similar when playing with API scripts with Development Center multiple times in a row.

Make sure you are not using ROS while Development Center. They are not compatible
You can run scripts that use the API while Development Center is running, but if you experience a similar issue (might happen after a couple of runs), restarting Development Center should fix the issue 
Finally, if the error still persist, I would try restarting the arm, and possibly the PC as well (but I don't think this will be necessary)

Best,
Felix

ericdusel77, 2022-05-24:
Hi Felix! Thank you for the quick response. 

I am unsure about this advice. I am not using both at the same time.  I was originally only using ROS, but noticed that it sometimes wouldn't boot up. After a couple tries it would be fine, but now it never stops blinking green so when I looked into that is when I discovered Development Center and updating the firmware. So they have never been "played with" at the _exact_ same time. I have also tried booting without being connected to any computer with no luck. I have powered everything off several times. 

In fact, DevCenter seems broken. It is version 1.5.1 and it doesn't seem to have any connection to the arm. All of the values under Update: Versions are 00.00.00, despite the connection being "Good" in Torque Controller and the firmware update saying it worked. It also says BOOTLOADER in the top right of DevCenter. DevCenter crashes a lot if I click anything except General Settings.. I have tried DevCenter on a Windows 10 machine and an Ubuntu 20.04 (I know it says 14.04 and 16.04 but it installed fine and behaves the same as the Windows 10 version).

[image]

[image]

[image]

[image]

[image]

I have tried two of the firmware. Uncertain which one was correct (maybe neither of them)

- FS 0CPP 0010_6.2.10_generic.hex
- FS 0CPP 0002_6.2.10_jaco_6dof_v2.hex

Finally, I followed the instructions #248 which mention that the OP did NOT short circuit the joystick, so I did not either since it did not come with a figure explaining how to do so.  

Arm serial number is PJ 0000 0001 0306 03 13004 0001

I literally just received this popup...
[image]

felixmaisonneuve, 2022-05-24:
Ok, I did not understand correctly previously so my first advice is not applicable here. 

I don't think DevCenter is broken. It seems your arm is stuck in the bootloader, which is weird. 
You tried 2 firmwares and they both should have worked, but the one you want to use is FS 0CPP 0002_6.2.10_jaco_6dof_v2.hex. 
The instructions you followed are correct.
To short the joystick, you need to connect these two holes (with a metal wire for example)
[image]

However, you need to short the joystick to boot into the bootloader, which you are already in, so I don't think this will be helpfull. 
You seem stuck inside the bootloader, meaning the firmware probably didn't flash correctly. I would have told you the flash the firmware again, but you already did that. You could try it again a couple times just to be sure, but I don't have much hope. 

The popup is normal. It simply compares the current "Main Firmware" version and compares it with what was read before the update. nothing to worry about here. 

I will recommend you to send an email to support@kinova.ca explaining the arm cannot boot correctly. It seems stuck inside the bootloader and updating the firmware doesn't seem to work (even if it says it succeeded) because you are still stuck with the flashing green light. They will probably send you the same procedure, juste tell them you already did that. 

I am not sure I can help you more than that since I do not know the protocol when there is a flashing issue.

Best of luck,
Felix

ericdusel77, 2022-05-26:
I had an old arm. I  spoke with support who got me a working firmware. All set.

Thank you Felix.

skpawar1305, 2023-04-24:
> I had an old arm. I  spoke with support who got me a working firmware. All set.
> 
> Thank you Felix.

hello, can you send this firmware file of you still have it on your mail or somewhere?
Kinova doesn't have anything of Jacov1 now. Thank you
my email is skpawar1305@gmail.com

felixmaisonneuve, 2023-05-10:
Hi @skpawar1305,
You can always contact support@kinova.ca for specific firmware demands.

Renzo-Mendoza, 2023-05-10:
Hi,
Same problem, I have an old Jaco arm (purchased by 2014) and I tried to boot both firmware from Kinova software web page, but the problem persists. Could you share that firmware by google drive? Do you have any firmware repository of Kinova arms by versions/year purchase/SN?

skpawar1305, 2023-05-11:
> Hi @skpawar1305,
> You can always contact support@kinova.ca for specific firmware demands.

yes, they shared, now some times it used to boot, not anymore 😆

Renzo-Mendoza, 2023-05-11:
> > Hi @skpawar1305,
> > You can always contact [support@kinova.ca](mailto:support@kinova.ca) for specific firmware demands.
> 
> yes, they shared, now some times it used to boot, not anymore laughing

Hello @skpawar1305,
Could you share that firmware? I load both newest firmware and my laptop cannot detect Jaco1.
Thanks!
My e-mail is: mendoza.rm@pucp.edu.pe

anc08, 2024-09-04:
> > > Hi @skpawar1305,
> > > You can always contact [support@kinova.ca](mailto:support@kinova.ca) for specific firmware demands.
> > 
> > 
> > yes, they shared, now some times it used to boot, not anymore laughing
> 
> Hello @skpawar1305, Could you share that firmware? I load both newest firmware and my laptop cannot detect Jaco1. Thanks! My e-mail is: [mendoza.rm@pucp.edu.pe](mailto:mendoza.rm@pucp.edu.pe)

Hi, same problem over here and kinova support is telling me I might have to send the arm into their headquarters to get it looked at, if anyone has the referenced older firmware could you send it to me so I can try that? Email me at cohenad3@msu.edu if so, thanks!

martinleroux, 2024-09-04:
Then reply to your support thread with this information and we will send you the appropriate firmware file. This would explain your symptoms.

anc08, 2024-09-06:
> Then reply to your support thread with this information and we will send you the appropriate firmware file. This would explain your symptoms.

Hello @martinleroux 

I gave that extra information to support and they did send me another firmware version to try. However, flashing that version caused my arm to get stuck in green flashing lights like the original post, and upon hearing that, support once again suggested sending the arm into Kinova HQ to be inspected. Do you have any other suggestions? I’m trying to avoid having to wait for the arm to be shipped and fixed if at all possible, since what I’m working on is somewhat time sensitive.

martinleroux, 2024-09-06:
I imagine support sent you a procedure called something like "bootloader procedure". You can try to run it with this new (old) firmware. If this doesn't work, then they are right and the robot will need to be shipped back to our office for inspection and repair.