Kinovarobotics/kinova-ros #392: The results of service: run_COM_parameters_estimation

yepw, 2022-06-12:
Hi, I was trying to use the `optimal` gravity mode on a j2s7s300 arm. To get the gravity parameters, I called service `/j2s7s300_driver/in/run_COM_parameters_estimation`. The robot automatically started moving. The calibration process finished after 10~15 minutes. 
However, the gravity parameters I got have **19** numbers. I couldn't come up with the meaning of the parameters, so I don't know how to use them. 
If there were 4 numbers, I would assume they are [COM, COMx, COMy, COMz]. If there were 28 numbers, I would assume they are [m1,...,m7,x1,...,x7,y1,...,y7,z1,...,z7]. 
The entire outputs are below. The same 19 numbers were saved to ParametersOptimal_Z.txt
```
[ INFO] [1654981878.464780392]: Running 7 dof robot COM estimation sequence
C A U T I O N : The robot will execute a trajectory. The user must remain alert and turn off the robot if something wrong occurs. 
Please read the HTML documentation before using this function.

Once completed, you will find the parameters in the file ParametersOptimal_Z.txt located in the same folder as your application.
0% 0% 1% 1% 2% 2% 3% 4% 4% 5% 5% 6% 6% 7% 8% 8% 9% 9% 10% 10% 11% 12% 12% 13% 13% 14% 14% 15% 16% 16% 17% 17% 18% 18% 19% 20% 20% 21% 21% 22% 22% 23% 24% 24% 25% 25% 26% 26% 27% 28% 28% 29% 29% 30% 30% 31% 32% 32% 33% 33% 34% 34% 35% 36% 36% 37% 37% 38% 38% 39% 40% 40% 41% 41% 42% 42% 43% 44% 44% 45% 45% 46% 46% 47% 48% 48% 49% 49% 50% 50% 51% 52% 52% 53% 53% 54% 54% 55% 56% 56% 57% 57% 58% 58% 59% 60% 60% 61% 61% 62% 62% 63% 64% 64% 65% 65% 66% 66% 67% 68% 68% 69% 69% 70% 70% 71% 72% 72% 73% 73% 74% 74% 75% 76% 76% 77% 77% 78% 78% 79% 80% 80% 81% 81% 82% 82% 83% 84% 84% 85% 85% 86% 86% 87% 88% 88% 89% 89% 90% 90% 91% 92% 92% 93% 93% 94% 94% 95% 96% 96% 97% 97% 98% 98% 99% 
The parameters are: 
Param[0] = -0.00969267
Param[1] = -1.55382
Param[2] = -0.00867128
Param[3] = -0.0373247
Param[4] = -0.0117275
Param[5] = -0.677411
Param[6] = -0.00295211
Param[7] = -0.00573624
Param[8] = -0.00193285
Param[9] = -0.23219
Param[10] = -0.000938933
Param[11] = -0.00253541
Param[12] = 0.0920915
Param[13] = 0.0879858
Param[14] = -0.342845
Param[15] = 0.419094
Param[16] = -0.195077
Param[17] = -0.0501154
Param[18] = 0.0739535

```

felixmaisonneuve, 2022-06-13:
Hi @yepw,

These values are the "optimal gravity parameters". For 6dof arms, there are 16 values that are detailed in the Documentation (see `SetGravityOptimalZParam()`). For 7dof arms, there are 19 values.

In our ROS driver, when calling `setRobotCOMParam`, you can choose the `GRAVITY_TYPE` between `MANUAL_INPUT` and `OPTIMAL`. 
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_comm.cpp#L851-L872

If you select `OPTIMAL`, `com_parameters` must have 19 values (for 7dof) and it will call the API function `SetGravityOptimalZParam()`. If you select `MANUAL_INPUT`, then it would require 28 values and it will call a different API function (`setGravityManualInputParam()`).

The `run_COM_parameters_estimation` service you called gives you the optimal gravity parameters and sets them (by calling `setGravityOptimalZParam`).

This configuration is reset to default everytime the arm is rebooted. However, you can edit your robot_parameters.yaml with your values so you won't have to do the whole 15minutes sequence again to change your arm's config.
To do so, change `use_estimated_COM_parameters` to `true`, uncomment `com_parameters` and add your values.
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_bringup/launch/config/robot_parameters.yaml#L25-L35

When this is done, all you have to do is launch the ROS driver (so the changes can take effect) and call the `in/set_torque_control_parameters` service
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_arm.cpp#L188-L190
It will call the `setTorqueControlParametersService` function, which will call `setRobotCOMParam` with the `OPTIMAL` parameter (based on your robot_parameters.yaml config)
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_arm.cpp#L294-L305

Regards,
Felix

yepw, 2022-06-15:
Thanks, @felixmaisonneuve! Gravity compensation works well with the 19 values.

closed 2022-06-15 by yepw