Kinovarobotics/kinova-ros #394: error occurs when using service /j2n6s300_driver/in/run_COM_parameters_estimation

zhangzc19, 2022-06-16:
## Description
when I try to use /j2n6s300_driver/in/run_COM_parameters_estimation to get COM parameters , I come up with the error:'transport error completing service call: unable to receive data from sender, check sender's logs for details'. 

Another command window's output is:

[ INFO] [1655382306.642208076]: Running COM estimation sequence
C A U T I O N : The robot will execute a trajectory. The user must remain alert and turn off the robot if something wrong occurs. 
Please read the HTML documentation before using this function.

Once completed, you will find the parameters in the file ParametersOptimal_Z.txt located in the same folder as your application.
terminate called after throwing an instance of 'kinova::KinovaCommException'
  what():  KinovaCommException: Could not get the angular command (return code: 1015)

[j2n6s300_driver-2] process has died [pid 1117837, exit code -6, cmd /home/stan/kinova_ws/devel/lib/kinova_driver/kinova_arm_driver j2n6s300 __name:=j2n6s300_driver __log:=/home/stan/.ros/log/56664710-ed6f-11ec-9b79-f1a22c0ca4ec/j2n6s300_driver-2.log].
log file: /home/stan/.ros/log/56664710-ed6f-11ec-9b79-f1a22c0ca4ec/j2n6s300_driver-2*.log

## Version

ROS distribution : 20.04

Branch and commit you are using : noetic-devel

ROBOTS: j2n6s300

## Steps to reproduce

1. roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300 use_urdf:=true 
 (i tried to use 'use_urdf:=false' but also failed )
2. rosservice call /j2n6s300_driver/in/home_arm   
(it works well)
3. rosrun kinova_demo run_COMParameters_estimation.py j2n6s300

hope to hear from you soon !

zhangzc19, 2022-06-17:
I‘m using USB connection in my own computer.I'm pretty sure that the physical connection works well(since there is no error with other service). But once I tried the ' /j2n6s300_driver/in/run_COM_parameters_estimation', the arm disconnects and then i cannot control the arm.
I don't install the fingers for my arm ,does it make any difference ?

felixmaisonneuve, 2022-06-17:
It might be, the actual error code 1015 is ERROR_NO_DEVICE_FOUND. This usually happens when connection is lost and the program cannot find the arm as a whole, but in your case it might be because a command is sent to the gripper and it cannot find it. I don't know the full run_COM_parameters_estimation sequence and I can't remember if a command is sent to the end-effector.

You should try to run it again with the gripper to see if it does the same error. However, if it works, the resulting parameters won't be optimal when you will remove the gripper afterwards.

zhangzc19, 2022-06-18:
Actually, I'm using my own end effector(a drill) instead of the fingers and I didn't buy the fingers. So is there any possibility that it can work well without the fingers?

felixmaisonneuve, 2022-06-20:
Just to make sure, when you call the service, your arm does not move at all and it crashes right away, is that correct?

The sequence sends 175 trajectory using angular points like this : 
```c++
TrajectoryPoint point;

//Initializing the point.
point.Limitations.accelerationParameter1 = 0.0f; //Not implemented yet but will be in a future release.
point.Limitations.accelerationParameter2 = 0.0f; //Not implemented yet but will be in a future release.
point.Limitations.accelerationParameter3 = 0.0f; //Not implemented yet but will be in a future release.
point.Limitations.forceParameter1 = 0.0f; //Not implemented yet but will be in a future release.
point.Limitations.forceParameter2 = 0.0f; //Not implemented yet but will be in a future release.
point.Limitations.forceParameter3 = 0.0f; //Not implemented yet but will be in a future release.

point.Limitations.speedParameter1 = 10;//We limit the translation velocity to 8 cm per second.
point.Limitations.speedParameter2 = 20; //We limit the orientation velocity to 0.6 RAD per second
point.Limitations.speedParameter3 = 10;

point.Position.Type = ANGULAR_POSITION;
point.Position.HandMode = HAND_NOMOVEMENT;

point.LimitationsActive = true;

// for each pose
{
	point.Position.Actuators.Actuator1 = 270; // changes for each pose
	point.Position.Actuators.Actuator2 = 180; // changes for each pose
	point.Position.Actuators.Actuator3 = 90; // changes for each pose
	point.Position.Actuators.Actuator4 = 270; // changes for each pose
	point.Position.Actuators.Actuator5 = 90; // changes for each pose
	point.Position.Actuators.Actuator6 = 90; // changes for each pose
	point.Position.Fingers.Finger1 = 6500;
	point.Position.Fingers.Finger2 = 6500;
	point.Position.Fingers.Finger3 = 6500;

	SendAdvanceTrajectory(point);
}
```

Can you do a quick test to see if you get the same error when calling `SendAdvanceTrajectory` with a similar `point`.
A quick way to do it would be to edit the `setJointAngles` function then call it.
https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_driver/src/kinova_comm.cpp#L553

To call it, I would edit the [homeArmServiceCallback](https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_arm.cpp#L244) function and simply call the /in/home_arm service
```c++
// kinova_comm_.homeArm();
// kinova_comm_.initFingers();
kinova_comm_.setJointAngles();
res.homearm_result = "KINOVA ARM HAS BEEN RETURNED HOME";
return true;
```

Make sure to do a `catkin_make` for the changes to take effect. 

I want to know if your error comes from sending these trajectories or if it comes from something else in the sequence.

Best,
Felix

felixmaisonneuve, 2022-08-15:
Hi @zhangzc19,

Did you try my suggestion? 
Was the finger command causing the issue?

I will assume it is the problem and close this issue in a couple weeks if I do not ear from you

Regards,
Felix

closed 2022-12-15 by felixmaisonneuve

raunaqbhirangi, 2023-12-05:
Hi, I have this same problem. I am also using a custom end effector and it seems like I get the error after the robot finishes homing, which seems to be the first step in run_COM_parameters_estimation. Is there a solution for this?