Kinovarobotics/kinova-ros #397: How to use multiple arms in ROS and Moveit?

Tzxtoiny, 2022-07-10:
Hello! I would like to ask how to use multiple arms in ROS and moveit?
I have read the Readme instructions but I am not able to make it in the first step.
# Things I have done
1. I have configured the "multiple_robots.yaml"
```
kinova_number_of_robots: 2
kinova_robots:
  - serial: PJ00900006526339-0
    type: j2s6s300
    name: left
  - serial: 12313
    type: j2s6s300
    name: right
```
The right arm serial number I have reset it to $12123$, it's ok!

2. I uncommented the line in the "kinova_robot,launch "
``  
	<rosparam file="$(find kinova_bringup)/launch/config/multiple_robots.yaml" command="load" />``

# ERRORS
Then I launched the driver of one of the arm which serial number is $12123$,
`roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s6s300 kinova_robotName:=right kinova_robotSerial:=12313
`
 the results is 
```
SUMMARY
========

PARAMETERS
 * /kinova_number_of_robots: 2
 * /kinova_robots: [{'serial': 'PJ00...
 * /right_driver/connection_type: USB
 * /right_driver/ethernet/local_broadcast_port: 25025
 * /right_driver/ethernet/local_cmd_port: 25000
 * /right_driver/ethernet/local_machine_IP: 192.168.100.100
 * /right_driver/ethernet/subnet_mask: 255.255.255.0
 * /right_driver/jointSpeedLimitParameter1: 10
 * /right_driver/jointSpeedLimitParameter2: 20
 * /right_driver/robot_name: right
 * /right_driver/robot_type: j2s6s300
 * /right_driver/serial_number: 12313
 * /right_driver/status_interval_seconds: 0.05
 * /right_driver/torque_parameters/publish_torque_with_gravity_compensation: True
 * /right_driver/torque_parameters/use_estimated_COM_parameters: False
 * /right_driver/use_jaco_v1_fingers: False
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.11

NODES
  /
    right_driver (kinova_driver/kinova_arm_driver)
    right_state_publisher (robot_state_publisher/robot_state_publisher)
    slow_pub (kinova_driver/slow_pub.py)

auto-starting new master
process[master]: started with pid [716049]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 9ef4d6e6-005b-11ed-a925-1bb2c2e46dfd
process[rosout-1]: started with pid [716060]
started core service [/rosout]
process[right_driver-2]: started with pid [716067]
process[right_state_publisher-3]: started with pid [716068]
process[slow_pub-4]: started with pid [716070]
[ INFO] [1657462903.995810849]: kinova_robotType is j2s6s300.
[ INFO] [1657462903.996831066]: kinova_robotName is right.
[ INFO] [1657462903.999562564]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ INFO] [1657462904.157230953]: Found 1 device(s), using device at index 0 (model: Spherical 6DOF Serv, serial number: 12313, code version: 393736, code revision: 0)
[ INFO] [1657462904.228227909]: Initializing fingers...this will take a few seconds and the fingers should open completely
**terminate called after throwing an instance of 'XmlRpc::XmlRpcException'
[right_driver-2] process has died [pid 716067, exit code -6, cmd /home/harlab/kinova_ws/devel/lib/kinova_driver/kinova_arm_driver j2s6s300 __name:=right_driver __log:=/home/harlab/.ros/log/9ef4d6e6-005b-11ed-a925-1bb2c2e46dfd/right_driver-2.log].
log file: /home/harlab/.ros/log/9ef4d6e6-005b-11ed-a925-1bb2c2e46dfd/right_driver-2*.log**
```

# Others 
when I commented the code
`<rosparam file="$(find kinova_bringup)/launch/config/multiple_robots.yaml" command="load" />`,
the I launched the arm
`roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s6s300 kinova_robotName:=right kinova_robotSerial:=12313
`
the error is gone , but there is a warning 
```
PARAMETERS
 * /right_driver/connection_type: USB
 * /right_driver/ethernet/local_broadcast_port: 25025
 * /right_driver/ethernet/local_cmd_port: 25000
 * /right_driver/ethernet/local_machine_IP: 192.168.100.100
 * /right_driver/ethernet/subnet_mask: 255.255.255.0
 * /right_driver/jointSpeedLimitParameter1: 10
 * /right_driver/jointSpeedLimitParameter2: 20
 * /right_driver/robot_name: right
 * /right_driver/robot_type: j2s6s300
 * /right_driver/serial_number: 12313
 * /right_driver/status_interval_seconds: 0.05
 * /right_driver/torque_parameters/publish_torque_with_gravity_compensation: True
 * /right_driver/torque_parameters/use_estimated_COM_parameters: False
 * /right_driver/use_jaco_v1_fingers: False
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.11

NODES
  /
    right_driver (kinova_driver/kinova_arm_driver)
    right_state_publisher (robot_state_publisher/robot_state_publisher)
    slow_pub (kinova_driver/slow_pub.py)

auto-starting new master
process[master]: started with pid [717218]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 1661a664-005c-11ed-a925-1bb2c2e46dfd
process[rosout-1]: started with pid [717229]
started core service [/rosout]
process[right_driver-2]: started with pid [717236]
process[right_state_publisher-3]: started with pid [717237]
process[slow_pub-4]: started with pid [717238]
[ INFO] [1657463104.313277089]: kinova_robotType is j2s6s300.
[ INFO] [1657463104.314163432]: kinova_robotName is right.
[ INFO] [1657463104.317076418]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ INFO] [1657463104.475913580]: Found 1 device(s), using device at index 0 (model: Spherical 6DOF Serv, serial number: 12313, code version: 393736, code revision: 0)
[ INFO] [1657463104.546907698]: Initializing fingers...this will take a few seconds and the fingers should open completely
[ INFO] [1657463104.621878247]: The arm is ready to use.
right_joint_1 right_joint_2 right_joint_3 right_joint_4 right_joint_5 right_joint_6 
[DEBUG] [1657463104.647855441]: Trying to publish message of type [control_msgs/FollowJointTrajectoryFeedback/10817c60c2486ef6b33e97dcd87f4474] on a publisher with type [control_msgs/FollowJointTrajectoryFeedback/10817c60c2486ef6b33e97dcd87f4474]
[DEBUG] [1657463104.682857928]: Trying to publish message of type [kinova_msgs/JointAngles/bf16c4e5154848a0767d1aaab0a4796b] on a publisher with type [kinova_msgs/JointAngles/bf16c4e5154848a0767d1aaab0a4796b]
[DEBUG] [1657463104.682911672]: Trying to publish message of type [sensor_msgs/JointState/3066dcd76a6cfaef579bd0f34173e9fd] on a publisher with type [sensor_msgs/JointState/3066dcd76a6cfaef579bd0f34173e9fd]
[DEBUG] [1657463104.686857898]: Trying to publish message of type [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5] on a publisher with type [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5]
[DEBUG] [1657463104.686892195]: Trying to publish message of type [kinova_msgs/KinovaPose/e831d993faea563f6fe69d7db9b384c9] on a publisher with type [kinova_msgs/KinovaPose/e831d993faea563f6fe69d7db9b384c9]
[DEBUG] [1657463104.688876325]: Trying to publish message of type [geometry_msgs/WrenchStamped/d78d3cb249ce23087ade7e7d0c40cfa7] on a publisher with type [geometry_msgs/WrenchStamped/d78d3cb249ce23087ade7e7d0c40cfa7]
[DEBUG] [1657463104.690835193]: Trying to publish message of type [kinova_msgs/FingerPosition/f56891e5dcd1900989f764a9b845c8e5] on a publisher with type [kinova_msgs/FingerPosition/f56891e5dcd1900989f764a9b845c8e5]
[DEBUG] [1657463104.845249047]: Trying to publish message of type [actionlib_msgs/GoalStatusArray/8b2b82f13216d0a8ea88bd3af735e619] on a publisher with type [actionlib_msgs/GoalStatusArray/8b2b82f13216d0a8ea88bd3af735e619]
**[ WARN] [1657463104.885910257]: Joint state with name: "right_joint_1" was received but not found in URDF**
```
I want to know how to fix this warning and the error above.

Please give some instructions on how to use multiple arms in ROS and in Moveit in detail, thank you very much. Are there any other files I need to modify?

felixmaisonneuve, 2022-07-11:
Hi @Tzxtoiny,

You could try to change `- serial: 12313` to `- serial: "12313"` in multiple_robots.yaml.
This yaml file is interpreted here : 
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_arm.cpp#L59-L63

This can be very sensitive to format and I think it considers your serial number as a number but it expects a string and it leads to an error. You could change the serial number to something with a letter and it should work as well.

I don't think this dual arm setup will work with moveit out of the box. It will probably require some modifications in the driver. I don't know exactly what it will take, but I can help you as problems come. 

For your warning, can you print the `/robot_description` rosparam? This will help us understand what is found in the urdf. 

Best,
Felix

Tzxtoiny, 2022-07-12:
> Hi @Tzxtoiny,
>
> You could try to change `- serial: 12313` to `- serial: "12313"` in multiple_robots.yaml. This yaml file is interpreted here : [...]

Hi,
Thanks for pointing out the error, the error is solved.
As for the warning I mentioned above, I think maybe it is because that the parameter "kinova_robotName:=right" did not pass to the parameter "prefix" in urdf. I modified the line in the j2s6s300.xacro
`   <xacro:macro name="j2s6s300" params="base_parent prefix:=j2s6s300"> `
to 
`   <xacro:macro name="j2s6s300" params="base_parent prefix:=right"> `

Then I launched the driver 
`roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s6s300 kinova_robotName:=right kinova_robotSerial:=12313`
The warning disappeared. 
BUT as for multiple arms, the parameter "prefix" in the urdf can not be two different at the same time, so how to fix this problem to use two arms at the same time?

felixmaisonneuve, 2022-12-15:
Hi @Tzxtoiny ,

Did you try my suggestion?
Did it work?

I will assume it is the problem and close this issue in a couple weeks if I do not ear from you

Regards,
Felix

closed 2024-05-01 by smoya23