Kinovarobotics/kinova-ros #399: Code: 1015 error and 'could not initialize Kinova API'

oscarjrc, 2022-07-22:
Every time I run the launch file so that the JACO can publish topics, it provides me with this error: "KinovaCommException: Could not initialize Kinova API (return code: 1015)" This launch file has worked before but recently it has been giving me this error and would not work no matter how many times I reset both the computer and the JACO. 

I understand that code:1015 is a connection issue as I have looked at possible solutions regarding the USB as ttyUSB0 doesn't appear and all solutions I've found online hasn't helped with this issues.

Any insight as to possible causes or fixes is greatly appreciated.

ROS distribution : Melodic

## Steps to reproduce

1. Run roslaunch kinova_brinup kinova_robot.launch

felixmaisonneuve, 2022-07-22:
Hi @oscarjrc,

Did you change anything in your ros driver and the launch files since it worked the last time?
If not, this might be more on the hardware side. Have you tried with a different usb cable and another usb port?

Regards,
Felix

oscarjrc, 2022-07-22:
I left the ros driver and the launch files the same before it stopped working. Yes, I've switched out the usb cable and the usb port. 

Thanks,
Oscar

felixmaisonneuve, 2022-07-25:
Are you using a particular setup (e.g. running ROS inside a VM)?
Can you paste your full output after your `roslaunch kinova_brinup kinova_robot.launch`?
Are you able to connect to the arm with Development Center?
If you cannot use the application, you might have a hardware issue (e.g. the USB connector in the base).

oscarjrc, 2022-07-25:
Thanks for the feedback, the issue was resolved after reflashing the firmware.

closed 2022-07-25 by oscarjrc