Kinovarobotics/kinova-ros #401: Errors with nodes and params in launch file of Kinova Jaco2 launch file (Beginner)

dpetrou00, 2022-08-05:
Hello, I am a fairly new programmer and completely new to ROS and am trying to externally control a Kinova Jaco2 robotic arm using the example Python scripts in the kinova-ros repository:

[https://github.com/Kinovarobotics/kin...](https://github.com/Kinovarobotics/kinova-ros)

My version of Ubuntu (20.04) uses ros-noetic per the GitHub page. In case I did something in the wrong order or am missing a step in my set-up, I will describe what I have done. After downloading ros-noetic on my machine, I sourced my environment with

`$ source /opt/ros/noetic/setup.bash`

and created a catkin_ws directory and src sub-directory and ran catkin_make. I then cloned the above repository into the src directory and ran catkin_make in the catkin_ws directory again. From there, I have played around with the roslaunch commands with multiple issues while looking at the .launch file for reference. In the repository, the .launch file is called "kinova_robot.launch" and it is in the folder "kinova_bringup." When I run

`$ roslaunch kinova_bringup kinova_robot.launch`

my output is as follows:

```
$ roslaunch kinova_bringup kinova_robot.launch
... logging to /home/demetriospetrou/.ros/log/b66994b0-14fb-11ed-8b4d-29740da4210c/roslaunch-79VNV42-11762.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: Invalid <param> tag: Cannot load command parameter [robot_description]: no such command [['/opt/ros/noetic/share/xacro/xacro.py', '/home/demetriospetrou/catkin_ws/src/kinova-ros/kinova_description/urdf/j2n6s300_standalone.xacro']]. 

Param xml is <param name="robot_description" command="$(find xacro)/xacro.py '$(find kinova_description)/urdf/$(arg kinova_robotType)_standalone.xacro'"/>
The traceback for the exception was written to the log file
```

I have also tried to set the argument "use_urdf" to false as I noticed that this was related to the parameters in the above issue from the .launch file and got the following:

```
$ roslaunch kinova_bringup kinova_robot.launch use_urdf:=false
... logging to /home/demetriospetrou/.ros/log/97477362-14fc-11ed-8b4d-29740da4210c/roslaunch-79VNV42-11824.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://79VNV42:44431/

SUMMARY
========

PARAMETERS
 * /j2n6s300_driver/connection_type: USB
 * /j2n6s300_driver/ethernet/local_broadcast_port: 25025
 * /j2n6s300_driver/ethernet/local_cmd_port: 25000
 * /j2n6s300_driver/ethernet/local_machine_IP: 192.168.100.100
 * /j2n6s300_driver/ethernet/subnet_mask: 255.255.255.0
 * /j2n6s300_driver/jointSpeedLimitParameter1: 10
 * /j2n6s300_driver/jointSpeedLimitParameter2: 20
 * /j2n6s300_driver/robot_name: j2n6s300
 * /j2n6s300_driver/robot_type: j2n6s300
 * /j2n6s300_driver/serial_number: not_set
 * /j2n6s300_driver/status_interval_seconds: 0.1
 * /j2n6s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /j2n6s300_driver/torque_parameters/use_estimated_COM_parameters: False
 * /j2n6s300_driver/use_jaco_v1_fingers: False
 * /j2n6s300_tf_updater/base_frame: root
 * /kinova_number_of_robots: 2
 * /kinova_robots: [{'serial': 'PJ00...
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    j2n6s300_driver (kinova_driver/kinova_arm_driver)
    j2n6s300_tf_updater (kinova_driver/kinova_tf_updater)

auto-starting new master
process[master]: started with pid [11832]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 97477362-14fc-11ed-8b4d-29740da4210c
process[rosout-1]: started with pid [11842]
started core service [/rosout]
ERROR: cannot launch node of type [kinova_driver/kinova_arm_driver]: Cannot locate node of type [kinova_arm_driver] in package [kinova_driver]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [kinova_driver/kinova_tf_updater]: Cannot locate node of type [kinova_tf_updater] in package [kinova_driver]. Make sure file exists in package path and permission is set to executable (chmod +x)
```

In both cases, I am confused what the issue is with my limited knowledge but it seems that some file is missing from the specified location. I believe once I can call "roslaunch" successfully, I can use the rosrun command to run scripts? Any help and guidance would be very appreciated. Thanks!

felixmaisonneuve, 2022-08-05:
Hi @dpetrou00,

I assume you also did a `source devel/setup.bash` in your catkin workspace after running `catkin_make`.

I think your error comes from the git repo you cloned. By default, it comes with the melodic-devel branch, but, since you are using ROS noetic, you want to use the noetic-devel branch. 
To do so, go to your kinova-ros repo
```
cd <your_catkin_ws>/src/kinova-ros
```

Then checkout the noetic-devel branch
```
git checkout noetic-devel
```

After that, run `catkin_make` in your catkin workspace and source devel/setup.bash again.

`roslaunch kinova_bringup kinova_robot.launch` assumes you are using a 6DOF Jaco2 arm non-spherical with a 3 fingers gripper, but you need to specify the robot type if it is not the case : `roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2n6s300`

If it does not work, please print the output of your `roslaunch` command

Regards
Felix

dpetrou00, 2022-08-05:
Thanks for your quick reply @felixmaisonneuve! I went ahead and followed your instructions and got one less error now.

```
$ roslaunch kinova_bringup kinova_robot.launch
... logging to /home/demetriospetrou/.ros/log/99f82650-1507-11ed-ac28-0b6adefe7423/roslaunch-79VNV42-3410.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://79VNV42:35387/

SUMMARY
========

PARAMETERS
 * /j2n6s300_driver/connection_type: USB
 * /j2n6s300_driver/ethernet/local_broadcast_port: 25025
 * /j2n6s300_driver/ethernet/local_cmd_port: 25000
 * /j2n6s300_driver/ethernet/local_machine_IP: 192.168.100.100
 * /j2n6s300_driver/ethernet/subnet_mask: 255.255.255.0
 * /j2n6s300_driver/jointSpeedLimitParameter1: 10
 * /j2n6s300_driver/jointSpeedLimitParameter2: 20
 * /j2n6s300_driver/robot_name: j2n6s300
 * /j2n6s300_driver/robot_type: j2n6s300
 * /j2n6s300_driver/serial_number: not_set
 * /j2n6s300_driver/status_interval_seconds: 0.1
 * /j2n6s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /j2n6s300_driver/torque_parameters/use_estimated_COM_parameters: False
 * /j2n6s300_driver/use_jaco_v1_fingers: False
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    j2n6s300_driver (kinova_driver/kinova_arm_driver)
    j2n6s300_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [3420]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 99f82650-1507-11ed-ac28-0b6adefe7423
process[rosout-1]: started with pid [3430]
started core service [/rosout]
ERROR: cannot launch node of type [kinova_driver/kinova_arm_driver]: Cannot locate node of type [kinova_arm_driver] in package [kinova_driver]. Make sure file exists in package path and permission is set to executable (chmod +x)
process[j2n6s300_state_publisher-3]: started with pid [3437]

```

To clarify, should I be using 'source devel/setup.bash' instead of 'source /opt/ros/noetic/setup.bash'? And when is each of them supposed to be used. Thanks again!

felixmaisonneuve, 2022-08-08:
I will answer you question first. 
You have to source both files everytime you want to use ROS. 
`source /opt/ros/noetic/setup.bash` will setup your ROS environnement. With this, you will be able to run commands like `roslaunch`, `catkin_make`, `rosservice`, `rosrun`...
This line can be added to your .bashrc so it is ran every time you launch a new terminal and you won't have to worry about it.

You also need to source your workspace's devel/setup.bash file `source  <your_catkin_ws_path>/devel/setup.bash`. This will make your workspace (that you compiled previously with `catkin_make`) available, so you will be able to run specific commands from your worspace (e.g. `roslaunch kinova_bringup kinova_robot.launch`). 
Usually, you want to source this file everytime you run `catkin_make` because your workspace might have changed.

This is briefly covered in the [Installing and Configuring Your ROS Environment Tutorial](http://wiki.ros.org/ROS/Tutorials/InstallingandConfiguringROSEnvironment)

The error you get says ROS cannot locate a node. I would run `catkin_make clean` (that cleans your workspace) and `catkin_make` (to compile everything in your workspace). Make sure you don't get errors during these. 
After that, make sure you correctly source your workspace devel/setup.bash file (`source devel/setup.bash`) and try again. 
I think it cannot locate the node because your workspace is not sourced correctly and it cannot find the executable

dpetrou00, 2022-08-09:
Hello Felix, I hope you had a great weekend and thanks for your ongoing help. I was finally able to run roslaunch without error after cleaning my workspace and working through a couple more errors. Sourcing files makes more sense now, thank you. It seems that my terminal is now getting stuck during a series of tests as it publishes some messages. It keeps freezing at actionlib_msgs. Any thoughts?

```
$ roslaunch kinova_bringup kinova_robot.launch
... logging to /home/demetriospetrou/.ros/log/d529a224-180b-11ed-b9d8-79d2e3593309/roslaunch-79VNV42-16483.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://79VNV42:33487/

SUMMARY
========

PARAMETERS
 * /j2n6s300_driver/connection_type: USB
 * /j2n6s300_driver/ethernet/local_broadcast_port: 25025
 * /j2n6s300_driver/ethernet/local_cmd_port: 25000
 * /j2n6s300_driver/ethernet/local_machine_IP: 192.168.100.100
 * /j2n6s300_driver/ethernet/subnet_mask: 255.255.255.0
 * /j2n6s300_driver/jointSpeedLimitParameter1: 10
 * /j2n6s300_driver/jointSpeedLimitParameter2: 20
 * /j2n6s300_driver/robot_name: j2n6s300
 * /j2n6s300_driver/robot_type: j2n6s300
 * /j2n6s300_driver/serial_number: not_set
 * /j2n6s300_driver/status_interval_seconds: 0.1
 * /j2n6s300_driver/torque_parameters/publish_torque_with_gravity_compensation: False
 * /j2n6s300_driver/torque_parameters/use_estimated_COM_parameters: False
 * /j2n6s300_driver/use_jaco_v1_fingers: False
 * /robot_description: <?xml version="1....
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    j2n6s300_driver (kinova_driver/kinova_arm_driver)
    j2n6s300_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [16493]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d529a224-180b-11ed-b9d8-79d2e3593309
process[rosout-1]: started with pid [16503]
started core service [/rosout]
process[j2n6s300_driver-2]: started with pid [16510]
process[j2n6s300_state_publisher-3]: started with pid [16511]
[ INFO] [1660067462.813016317]: kinova_robotType is j2n6s300.
[ INFO] [1660067462.813997931]: kinova_robotName is j2n6s300.
[ INFO] [1660067462.817209921]: Initializing Kinova USB API (header version: 50300, library version: 5.2.0)
[ INFO] [1660067462.975299564]: Found 1 device(s), using device at index 0 (model: Jaco V2 Service    , serial number: PJ00900006535584-0 , code version: 393736, code revision: 0)
[ INFO] [1660067463.047242127]: Initializing fingers...this will take a few seconds and the fingers should open completely
[ INFO] [1660067463.137397953]: The arm is ready to use.
j2n6s300_joint_1 j2n6s300_joint_2 j2n6s300_joint_3 j2n6s300_joint_4 j2n6s300_joint_5 j2n6s300_joint_6 
[DEBUG] [1660067463.172174509]: Trying to publish message of type [control_msgs/FollowJointTrajectoryFeedback/10817c60c2486ef6b33e97dcd87f4474] on a publisher with type [control_msgs/FollowJointTrajectoryFeedback/10817c60c2486ef6b33e97dcd87f4474]
[DEBUG] [1660067463.249406895]: Trying to publish message of type [kinova_msgs/JointAngles/bf16c4e5154848a0767d1aaab0a4796b] on a publisher with type [kinova_msgs/JointAngles/bf16c4e5154848a0767d1aaab0a4796b]
[DEBUG] [1660067463.249557312]: Trying to publish message of type [sensor_msgs/JointState/3066dcd76a6cfaef579bd0f34173e9fd] on a publisher with type [sensor_msgs/JointState/3066dcd76a6cfaef579bd0f34173e9fd]
[DEBUG] [1660067463.253279399]: Trying to publish message of type [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5] on a publisher with type [geometry_msgs/PoseStamped/d3812c3cbc69362b77dc0b19b345f8f5]
[DEBUG] [1660067463.253361515]: Trying to publish message of type [kinova_msgs/KinovaPose/e831d993faea563f6fe69d7db9b384c9] on a publisher with type [kinova_msgs/KinovaPose/e831d993faea563f6fe69d7db9b384c9]
[DEBUG] [1660067463.255246946]: Trying to publish message of type [geometry_msgs/WrenchStamped/d78d3cb249ce23087ade7e7d0c40cfa7] on a publisher with type [geometry_msgs/WrenchStamped/d78d3cb249ce23087ade7e7d0c40cfa7]
[DEBUG] [1660067463.257374367]: Trying to publish message of type [kinova_msgs/FingerPosition/f56891e5dcd1900989f764a9b845c8e5] on a publisher with type [kinova_msgs/FingerPosition/f56891e5dcd1900989f764a9b845c8e5]
[DEBUG] [1660067463.357332390]: Trying to publish message of type [actionlib_msgs/GoalStatusArray/8b2b82f13216d0a8ea88bd3af735e619] on a publisher with type [actionlib_msgs/GoalStatusArray/8b2b82f13216d0a8ea88bd3af735e619]
```

felixmaisonneuve, 2022-08-09:
The error code 1015 is a connection issue. 
A couple questions : 

Are you using a USB or an Ethernet cable to connect to the arm? By default, our ROS stack uses the USB connection.
Have you installed the USB driver? See the note about the udev rule in the main [readme](https://github.com/Kinovarobotics/kinova-ros#installation)
Are you able to use the Development Center application to control the arm?
What arm are you using? By default, our ROS stack tries to connect to a 6dof (6 joints) arm with a non-spherical wrist and a 3 fingers gripper.

dpetrou00, 2022-08-09:
1. I am using a USB cable and I confirmed that the robot_parameters.yaml file still indicates connection_type as USB.
2. I manually copied and pasted the 10-kinova-arm.rules file into /etc/udev/rules.d. Not sure if something else is needed.
3. Yes, I have tested controlling the arm with the Development Center and can successfully do it.
4. My arm is 7 DOF with a 3-finger gripper and I believe a spherical wrist? I will attach a picture to confirm. Would the correct code for this be j2s7s300? Also, what is service vs. assistive? I ran roslaunch with this kinova_robotType and it seemed to work as the fingers opened, but it still froze on the same issue above.

Also, I noticed that the joystick and Development Center did not work after I executed roslaunch. Can the arm only be controlled by these or the API or can I control it simultaneously somehow. Thanks!

felixmaisonneuve, 2022-08-09:
thank you for your answer. 1, 2 and 3 are perfect.

For 4, yes, you will need to specify your specific robotType when launching your driver : `roslaunch kinova_bringup kinova_robot.launch kinova_robotType:=j2s7s300` or change the default option in your launch file : 
https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_bringup/launch/kinova_robot.launch#L4

All 7dof arms have spherical wrists, so j2s7s300 is good. 

Arms can be configured in Service or Assistive mode. Assistive mode prevents you from using certain features (e.g. torque control mode). 7Dof arms however are only available in service mode, so you shouldn't have to worry about it.

You cannot use the API/DevCenter and ROS at the same time. One of them (or both) will not work or it could create weird behaviour/errors. It is recommanded to only use one at the time.  

Since your are able to launch correctly and it crashes only when doing the `GetAngularCommand` : 
https://github.com/Kinovarobotics/kinova-ros/blob/93f66822ec073827eac5de59ebc2a3c5f06bf17f/kinova_driver/src/kinova_comm.cpp#L515-L524

Which is called here : 
https://github.com/Kinovarobotics/kinova-ros/blob/melodic-devel/kinova_driver/src/kinova_arm.cpp#L585

I would guess you either have an older firmware that (somehow) doesn't have this API fonction available, so it crashes, or you are using an older version of the API.

What is the firmware's version on your arm? You can check this in the "Update" page of the "General Settings" window in Development Center. 
What branch/commit of the kinova-ros repo are you using? Because Development Center is working correctly, maybe you are using and older commit of our kinova-ros repo that doesn't have the most recent version of the API.

Best,
Felix

dpetrou00, 2022-08-10:
Hello Felix, I just checked the firmware and saw that it is 6.2.8.0 while the most recent version is 6.2.10 on the Kinova website. Also, I noticed my arm is 6DOF and not 7DOF after looking at the settings like you mentioned. Since my desktop has Ubuntu 20.04, I am using the noetic-devel branch. I downloaded the 6.2.10 firmware hex file from the website, but I am not sure how to get it on my robotic arm and did not find instructions. Could you help me with that? I will also attach a picture of my arm to confirm it is 6DOF and spherical with you. Finally, do you have any suggestions to create shared control from two inputs since ROS and joystick combination is not possible? I am thinking of taking in keyboard input and some other input using the ROS and doing it that way, but can you think of some other solution?

[image]

felixmaisonneuve, 2022-08-11:
This is a 6DOF non-spherical (j2n6s300), so the default parameter in the launch file is correct and you can juste use `roslaunch kinova_bringup kinova_robot.launch`.

The noetic-devel branch is fairly new, so it shouldn't be the issue. By default it clones the melodic-devel branch, but if you did a `git checkout noetic-devel` and compiled everything afterwards with `catkin_make` (which you already did), you are all set.

For your arm's firmware, there is small changes between 6.2.8 and 6.2.10, but the API should be the same in both versions. However, if there was a bug or something, flashing a new firmware could fix the issue.
To install 6.2.10, it is very easy. You go in DevCenter, you open the "General Settings" page, you select the "Update" tab, you click "browse" and you select your .hex file (make sure you downloaded the correct one for your arm -> 6DOF non-spherical) and you click "Update". You have to wait a minute or two, then a popup will appear. You click "OK" then reboot the arm. Once your arm connects back to DevCenter, your arm's firmware version should have changed.

It is safe to use the Joystick at the same time as ROS or as DevCenter. Problem occurs when running ROS at the same time as DevCenter because they both use the same API to communicate with the arm, so it can create conflicts in the communication. The Joystick uses another method of communication entirely so there is nothing to worry about. You can use your Joystick controller safely at any moment. 

If updating the firmware does not work. You could try to "Factory Reset" the arm. The option is in the same "Update" tab. 

Regards,
Felix

felixmaisonneuve, 2022-12-15:
Hi @dpetrou00,

Did you manage to resolve the issue?
Can we close this?

I will close the issue regardless in a couple of days/weeks if I do not get any update.

Best,
Felix

smoya23, 2024-05-01:
Closing this issue as it has been inactive for a while and we got no further feedback

closed 2024-05-01 by smoya23