Kinovarobotics/kinova-ros #402: Cartesian angular velocity is small

Wang-Ruohan, 2022-09-23:
Hello, 
I'm seeking for the method to turn up the speed for my kinova arm. 
My kinova arm is KINOVA JACO2 (6 DoF, Spherical);
My ROS system is ROS kinetic.
I have two same type of kinova arms. Both are set to the maximum cartesian velocity of 0.2m/s using the software offered by kinova.
They all move well in cartesian position velocity control by using the topic command:
```
rostopic pub -r 100 /right_arm_driver/in/joint_velocity kinova_msgs/JointVelocity "{joint1: 0.0, joint2: 0.0, joint3: 0.0, joint4: 0.0, joint5: 0.0, joint6: 10.0, joint7: 0.0}"
```
Meaning that joint6 is rotating at the speed of 10 degrees/s.
However, when i send :
```
rostopic pub -r 100 /right_arm_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 0.0, twist_linear_y: 0.0, twist_linear_z: 0.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 10.0}"
```
to the right arm, and:
```
rostopic pub -r 100 /left_arm_driver/in/cartesian_velocity kinova_msgs/PoseVelocity "{twist_linear_x: 0.0, twist_linear_y: 0.0, twist_linear_z: 0.0, twist_angular_x: 0.0, twist_angular_y: 0.0, twist_angular_z: 10.0}"
```
to the left arm.
The joint6 of left arm moves at a significantly slower speed compaired to right arm. Note that I have set the two arms in a same pose.
Also, I have tried to operated the arm via joystick, it have the same performance as it mentioned above.
I have tried to update the firmware, but it didn't help.
Any help and guidance would be very appreciated. Thanks!

felixmaisonneuve, 2022-09-28:
Hi @Wang-Ruohan,

Can you compare the velocity limits from the Monitoring page of the Development Center app?
Maybe you could send a screenshot as well
[image]

Note: All my values are 0 because I am not connected to an arm.

Best,
Felix

Wang-Ruohan, 2022-10-13:
Thank you for your kind suggestion. I checked my virtual joystick setting and finally found the cartesian orientation velocity is small. When i turn the number up, the problem solved.

felixmaisonneuve, 2022-10-13:
Great,
I will close this issue then.

Best Regards,
Felix

closed 2022-10-13 by felixmaisonneuve