Kinovarobotics/kinova-ros #413: Gravity Compensation for Sideways Mount

zulfiqar-zaidi, 2023-01-19:
Hi,

I have a Kinovo j2s7s300 arm mounted sideways (base plate orthogonal to floor). I am using noetic-devel branch with Ubuntu 20. I have updated the urdf to reflect the mounting position w.r.t world (see attached picture). I want to use gravity compensation so the users can move the arm around freely. I had following questions:

- Is updating the urdf enough for the gravity compensation to work properly?
- I found older posts related to this. It was suggested to update the gravity vector. Would this still be required given that the urdf is updated? Secondly, the links in that answer don't work anymore. Could you re-post the link to documentation that talks about updating the gravity vector.
- If I move the arm to candle position and run `rosservice call /'${kinova_robotType}_driver'/in/set_zero_torques`, would that be a valid way  to do the gravity calibration?
- Any other suggestions or things that I need to be wary of.

Thank you!

[image]

felixmaisonneuve, 2023-01-20:
Hi @zulfiqar-zaidi,

I found the issue #164 you are talking about. 
I think the answer still applies and you still need to change the gravity vector using the Torque Console. The instructions are detailed in the [SDK User guide](https://drive.google.com/file/d/1y5TByFsF97s4_zh14E-q0YHSlMr5P1qe/view?usp=sharing) (p.40, section 5) available on our website in the [Resources](https://www.kinovarobotics.com/resources?r=339&s) section.

We provide HTML documentation for the API with the SDK software. This documentation is (normally) located in /opt/JACO-SDK/Documentation/OfflineDocumentation.html. You can search for `SetGravityVector` there and get some more info.

Changing the urdf will be enough to use moveit (I think only MoveIt/Gazebo uses URDF files), but you will need to chage the gravity vector to use cartesian velocity commands for example. 

Doing a `set_zero_torque`when the arm is not mounted vertically is not ideal. I think the best position to reset torques when mounted horizontally would be to make you rotate your second actuator to make the arm point upwards. You want to reduce the torque induced by gravity. You want the arm to be as vertical as possible

I don't have any other suggestion other then changing the gravity vector. It is not very common (for me anyways) to work with arms mounted sideways, so I don't know how well the arms react in such conditions. 
If you notice a problem and you think it could be related to this, make sure to shared it and we will see what we can do. 

For now, I think it should be enough to get you started

Best Regards,
Felix

smoya23, 2024-05-01:
Closing this issue as answer was provided and issue has been inactive for a while, don't hesitate to create a new one if you have any other questions.

closed 2024-05-01 by smoya23