Kinovarobotics/kinova-ros #417: ik fast

lixryjz, 2023-02-02:
hello ，I see there are three ikfast file，could you tell us ，what file is suitable to what type of robot arm and graspper？
Thanks！

felixmaisonneuve, 2023-02-06:
Hi @lixryjz,

Are you talking about 
kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/include/ikfast.h
kinova_moveit/inverse_kinematics_plugins/ikfast/j2s6s300_ikfast/include/ikfast.h
kinova_moveit/inverse_kinematics_plugins/ikfast/j2s7s300_ikfast/include/ikfast.h

From what I see, every IKfast file is within a folder for each arm config. The IK is computed in a <arm_config>_arm_ikfast_solver.cpp
example : kinova_moveit/inverse_kinematics_plugins/ikfast/j2n6s300_ikfast/src/j2n6s300_arm_ikfast_solver.cpp

and you can find an IK solver for j2n6s300, j2s6s300, j2s7s300 and m1n6s300

https://github.com/Kinovarobotics/kinova-ros/tree/melodic-devel/kinova_moveit/inverse_kinematics_plugins/ikfast

Is this what you are looking for?
If not, I may not understand your question correctly. Can you detail a bit more?

Best,
Felix

lixryjz, 2023-02-07:
Hi,thanks for your reply!As you know,the 7dof arm can use different graspper,like the followed figure,I hope to know the ikfast file is only for the arm ik to solve,or with a kind of graspper?,and I need to use the ikfast file to compile without any main function as a shared object,so I hope to know what file is refer to what kind of arm and endefferctor?The problem is the endeffector
[image]

lixryjz, 2023-02-07:
the ikfast file should  use without grasppe?and should I convert the coodinate?

lixryjz, 2023-02-07:
for this question，I don`t know the endeffector is what which is to build the ikfast file,so i  don`t know the file result  is reparent what, so i hope to know the the endeffector in this ikfast file!Thanks!

felixmaisonneuve, 2023-02-07:
Ok, 

The j2s7s300_arm_ikfast_solver.cpp file is for the jaco2 arm (j2), with a spherical wrist (s), 7dof (7), service mode (s) and a 3 fingers gripper (3). 

Is this what you want to know?

lixryjz, 2023-02-08:
yes，thank you very much！But is there an ikfast file for j2s7_arm which without the gripper?

[image]

felixmaisonneuve, 2023-02-13:
Looking deeper into this, I am pretty sure the ikfast_solver.cpp file is generated by the update_ikfast_plugin.sh script (that you can find in the arm's folder). It calls [create_ikfast_moveit_plugin.py](https://github.com/ros-planning/moveit_ikfast/blob/kinetic-devel/scripts/create_ikfast_moveit_plugin.py). Arguments for this script are `robot_name`, `planning_group_name`, `moveit_plugin_pkg` and `ikfast_output_file`. I see we are always using `arm` as the `planning_group_name`, so I think the ikfast file is only for the section of the arm without the gripper.

This means you could use the ikfast file for an arm without any gripper and it should work

closed 2024-05-01 by smoya23