Kinovarobotics/kinova-ros #420: Tool position in gazebo

rsaif22, 2023-05-11:
Hello,
I was trying to get the end effector trajectory from both the robot and the gazebo simulation. For the robot, this is published on the topic: /j2n6s300_driver/out/tool_pose. For the Gazebo simulation, however, this topic does not exist. In trying to get the position, I am using the tf transform from /root to /j2n6s300_end_effector, then taking the x and y positions from there. 

The trajectories do not match, as seen in the image:
[image]

Is this the correct approach to getting the tool position in Gazebo? 

Thank you

rsaif22, 2023-05-12:
Following up on this, do the Gazebo joint angles accurately mirror the real life joint angles? If not, is there an offset that can be applied?

xmaww, 2023-12-27:
Hey @rsaif22 ! May I ask if your problem has been solved? I faced the same question that I want to obtain both the tool_pose information and cartesian_client_control API in gazebo simulation, which can be easily accessible with real Kinova jaco robots. Thank you so much if you can reply!