Kinovarobotics/kinova-ros #422: Jaco1: Problems with joint control and toque control mode

Renzo-Mendoza, 2023-05-25:
Hello,

I am currently working with the Jaco1 and its firmware version 5.1.0 with ROS Noetic on Ubuntu 20.04. and I would appreciate some help with a couple of problems.

I was testing the joints angular control and I sent the arm at home position and registered these values, then I tried to place the arm to a position close to this with the following command:
_$ rosrun kinova_demo joints_action_client.py -v -r j2n6s300 degree -- 279 170 50 226 93 80_
However, the arm moved to a different position with the following values: 
_jnt1 - 307.8, jnt2 - 440.8, jnt3 - 151.4, jnt4 - 313.9, jnt5 - 278.8, jnt6 - 240.3_
What could be happening?

Also for some reason I can't get into torque control mode as it does not respond to the torque commands I send. I have tried the following options with no results:
- Enabling the mode via terminal and setting torque control parameters (safety factor = 1.0).
- I tried running the gravity compensation code but it does not set the motors torque to zero and stalls after activating it. 
- I tried using the Developer Center to activate the admittance control but it didn't work.
- I have tried through the Torque Control software but that doesn't work either.
I would like to know if Jaco1 supports torque control and if so I would appreciate some help with this problem.

Thanks!

Renzo-Mendoza, 2023-05-25:
Oh, I see I was using relative angles for the first problem.

felixmaisonneuve, 2023-05-25:
For the second issue with torque control, this mode is not available on Jaco1 arms unfortunately

Renzo-Mendoza, 2023-05-25:
> For the second issue with torque control, this mode is not available on Jaco1 arms unfortunately

Is admittance control not available?

felixmaisonneuve, 2023-05-26:
I have confirmed that you cannot switch to torque control mode (admittance control) on a Jaco1 arm.
These are very old, I don't even know if you can upgrade to a more recent firmware than 5.1.0

Renzo-Mendoza, 2023-05-27:
Oh I see, Is there another way to send torque comands?

felixmaisonneuve, 2023-05-30:
I don't think so. Torque control is simply not supported on Jaco1 arms

Sorry for the inconvenience

closed 2023-05-31 by felixmaisonneuve