Kinovarobotics/kinova-ros #423: Torque control frequency

skywoodsz, 2023-06-10:
Hi, I want to know what is the maximum frequency of the torque command. The maximum is 100HZ or higher？
Meanwhile, what is the maximum feedback frequency of joint position, speed, and torque measurement values, if I choose to call the api function directly.
Thanks!

felixmaisonneuve, 2023-06-16:
Hi @skywoodsz,

The main control loop of the Jaco2 base is running at 100Hz. So the API, and therefore ROS (because ROS is running the API in the background) will be limited at 100Hz as well.

If you absolutely need to go faster, we can provide the communication protocol used in the arm so you can send commands directly to each actuators (but then you loose all the functionnalities of the base, e.g. Kinematics). This can help you increase to 300-500Hz, but it makes everything way more difficult. 

Best,
Felix

skywoodsz, 2023-06-18:
> 

Thanks for your reply!
How can I get the communication protocol and its documentation? 
I also found that the presence of friction in torque control increases the error.
Is there any compensation for friction inside the base? Thanks！

skywoodsz, 2023-06-18:
> Hi @skywoodsz,
>
> The main control loop of the Jaco2 base is running at 100Hz. So the API, and therefore ROS (because ROS is running the API in the background) will be limited at 100Hz as well.
>
> If [...]

Thanks for your reply!
How can I get the communication protocol and its documentation?
I also found that the presence of friction in torque control increases the error.
Is there any compensation for friction inside the base? Thanks！

felixmaisonneuve, 2023-06-26:
There is no compensation for torque friction. Torque tracking is no really on point on Jaco2 arms. This arm was not designed for such advanced use cases, so it not super precise and accurate (good enough for most use cases, but not ideal for more advance work).

You can request the Low-Level API documentation from support@kinova.ca 

Best,
Felix

skywoodsz, 2023-06-27:
> There is no compensation for torque friction. Torque tracking is no really on point on Jaco2 arms. This arm was not designed for such advanced use cases, so it not super precise and accurate (good enough for most use cases, but not ideal for more advance work).
> 
> You can request the Low-Level API documentation from [support@kinova.ca](mailto:support@kinova.ca)
> 
> Best, Felix

Thanks for your reply!
I will do that.

closed 2023-06-27 by skywoodsz