Kinovarobotics/kinova-ros #425: start_force_control_service on j2s7s300 rotates J7/wrist to 180 radians and then locks J7 in that position during force control

katallen405, 2023-07-06:
## Description
When running the start_force_control service on the j2s7s300, the wrist rotates to 180 *radians* (so it takes several minutes) and then stays locked in that position.  

## Version

ROS distribution : noetic (though we also had this issue in melodic)

Branch and commit you are using : noetic-devel

## Steps to reproduce

1. roslaunch the arm
2. rosservice call /j2s7s300_driver/in/start_force_control 

## Expected behavior

Expected behavior is for joint 7 to remain at its current position and to be compliant.

Arm is mounted vertically

felixmaisonneuve, 2023-07-07:
Hi @katallen405,

Have you tried to reset torques before you start_force_control?
Before calling start_force_control, what are yor torques reading for each actuator?

Best,
Felix

katallen405, 2023-07-07:
What do you mean by reset torques? (So I think the answer to your question is no!)Is that a service I would call?Peace,Kat Allen

katallen405, 2023-07-07:
> Torque sensors drift over time and one actuator might read a torque value even when no force is applied. This can result in un wanted movements when switching to admittance control.
> Procedure on how to reset torque is detailled in the User Guide in the Advacned section. You can use Development Center to reset torque values or you can call the /<arm>_driver/in/set_zero_torques service.
> Normally, you'll want to put the in a candle position facing up (all joints to 180deg). However, since you are mounted vertically (and by that I assume arm is mounted on a wall), your second actuator to 90deg or 270deg so the arm is either facing up or down. The goal is to put the arm in a position where no torque is applied on the actuator. It is not recommanded to reset torque values if not mounted on a leveled surface (ground or table), but I think it should work nonetheless
> Also, since you are mounted vertically, did you change the gravity vector of the arm?
> Or maybe you are mounted normally and none of this applies to you.

Thanks! We were able to put the arm in “candlestick mode” while connected to the Windows driver/software and reset the torques and that did fix the issue - thank you so much. We’d tried to fix it within ROS and for some reason it didn’t seem to be working correctly, but now it works in ROS also.Thanks so much for the help!Kat Allen

felixmaisonneuve, 2023-07-10:
Great news,

I will close this issue then.

Regards,
Felix

closed 2023-07-10 by felixmaisonneuve