Kinovarobotics/kinova-ros #426: com parameters on j2s7s300

katallen405, 2023-07-06:
## Description

Is there documentation somewhere for the values and effect of the com parameters set by rosservice call /j2s7s300_driver/in/set_force_control_params "inertia_linear: {x: 0.0, y: 0.0, z: 0.0}
inertia_angular: {x: 0.0, y: 0.0, z: 0.0}
damping_linear: {x: 0.0, y: 0.0, z: 0.0}
damping_angular: {x: 0.0, y: 0.0, z: 0.0}
force_min_linear: {x: 0.0, y: 0.0, z: 0.0}
force_min_angular: {x: 0.0, y: 0.0, z: 0.0}
force_max_linear: {x: 0.0, y: 0.0, z: 0.0}
force_max_angular: {x: 0.0, y: 0.0, z: 0.0}"

## Version

ROS distribution : noetic-devel

Branch and commit you are using : 

## Details

I can run the /j2s7s300_driver/in/run_COM_parameters_estimation service, but the output does not match the inputs required for set_force_control_params service.  What do these numbers mean, and how to they relate? 
I want to reduce the force required to move the arm in force control mode, but changing the parameters blindly does not seem to have changed the responsiveness, and it is unclear what units the values are in.

felixmaisonneuve, 2023-07-07:
Hi @katallen405,

You can see what the rosservice /j2s7s300_driver/in/set_force_control_params does here :
https://github.com/Kinovarobotics/kinova-ros/blob/d8a6b5a1434bdaed9b16fa7a14965ca8a1b8a998/kinova_driver/src/kinova_arm.cpp#L174

https://github.com/Kinovarobotics/kinova-ros/blob/d8a6b5a1434bdaed9b16fa7a14965ca8a1b8a998/kinova_driver/src/kinova_arm.cpp#L409-L445

You see that it calls `setCartesianInertiaDamping` and `setCartesianForceMinMax`.

If you have installed the KinovaSDK 1.5.1 (available on our website), it provides html documentation. On Ubuntu, the default installation path is /opt/JACO-SDK/Documentation/OfflineDocumentation.html.

From there, you can search the API calls you want.

Force limits, damping, and inertia are pretty self explanatory, but basically they will affect cartesian admittance like this : 
`acceleration = (force_limits - dampin * velocity) / inertia`

COM parameters are a bit more complicated. I would suggest to read the `SetGravityOptimalZParam` documentation page for a detailed explanation of what each parameter does.

Hope this helps,
Felix

smoya23, 2024-05-01:
Closing this issue as answer was provided and issue has not been active for a while. Don't hesitate to open a new one for any other questions

closed 2024-05-01 by smoya23