Kinovarobotics/kinova-ros #442: URDF mass does not match spec sheet

tmckenna-1, 2024-07-18:
I am trying to do some center of mass and inertial calculations with the kinova arm. The spec sheet for the Jaco Gen2 7DoF spherical arm lists the mass (with 3 finger gripper) as 6.18 kg.

However, adding up the link masses from the urdf (j2s7s300.xacro) gets you a mass of 5.23 kg.

    <xacro:property name="link_base_mesh" value="base" />
    <xacro:property name="link_1_mesh" value="shoulder" />
    <xacro:property name="link_2_mesh" value="arm_half_1" />
    <xacro:property name="link_3_mesh" value="arm_half_2" />
    <xacro:property name="link_4_mesh" value="forearm" />
    <xacro:property name="link_5_mesh" value="wrist_spherical_1" />
    <xacro:property name="link_6_mesh" value="wrist_spherical_2" />
    <xacro:property name="link_7_mesh" value="hand_3finger" />

Is there something I am missing here? Is the urdf incorrect? Are the joint motor masses included in the urdf link masses?

martinleroux, 2024-07-19:
Hi @tmckenna-1 ,

You are correct, there seem to be an error in the .xacro file - the correct values are the ones from the user guide. Just looking at the numbers quickly, I can't quite figure out what went wrong, but I was able to confirm that our models in Matlab do sum up to the correct masses.

I'll add to my TODO list to correct the values for all of the models when I have time, but in the meantime, here is the URDF for j2s7s300.

```
<?xml version="1.0" encoding="UTF-8"?>
<robot name="Robot">
    <link name="j2s7s300_end_effector">
        <inertial>
            <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.00000000 0.00000000 0.00000000"/>
            <mass value="0.00000000"/>
            <inertia ixx="0.00000000" ixy="0.00000000" ixz="0.00000000" iyy="0.00000000" iyz="0.00000000" izz="0.00000000"/>
        </inertial>
        <visual>
            <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.00000000 0.00000000 0.00000000"/>
            <geometry>
                <box size="0.00000000 0.00000000 0.00000000"/>
            </geometry>
            <material name="">
                <color rgba="0.50000000 0.50000000 0.50000000 1.00000000"/>
            </material>
        </visual>
        <collision>
            <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.00000000 0.00000000 0.00000000"/>
            <geometry>
                <box size="0.00000000 0.00000000 0.00000000"/>
            </geometry>
        </collision>
    </link>
    <link name="j2s7s300_link_1">
        <inertial>
            <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.00000000 -0.00200000 -0.06050000"/>
            <mass value="0.74770000"/>
            <inertia ixx="0.00152032" ixy="0.00000000" ixz="0.00000000" iyy="0.00152032" iyz="0.00000000" izz="0.00059816"/>
        </inertial>
        <visual>
            <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.00000000 0.00000000 0.00000000"/>
            <geometry>
                <mesh filename="meshes\visual\shoulder.STL" scale="1.00000000 1.00000000 1.00000000"/>
            </geometry>
            <material name="">
                <color rgba="0.79215686 0.81960784 0.93333333 1.00000000"/>
            </material>
        </visual>
        <visual>
            <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.00000000 0.00000000 0.00000000"/>
            <geometry>
                <mesh filename="meshes\visual\ring_big.STL" scale="1.00000000 1.00000000 1.00000000"/>
            </geometry>
            <material name="">
                <color rgba="0.50000000 0.50000000 0.50000000 1.00000000"/>
            </material>
        </visual>
        <collision>
            <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.00000000 0.00000000 0.00000000"/>
            <geometry>
                <mesh filename="meshes\collision\shoulder.STL" scale="1.00000000 1.00000000 1.00000000"/>
            </geometry>
        </collision>
    </link>
    <link name="j2s7s300_link_2">
        <inertial>
            <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.00000000 -0.10356321  0.00000000"/>
            <mass value="0.84470000"/>
            <inertia ixx="0.00247074" ixy="0.00000000" ixz="0.00000000" iyy="0.00038011" iyz="0.00000000" izz="0.00247074"/>
        </inertial>
        <visual>
            <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.00000000 0.00000000 0.00000000"/>
[... 444 lines cut ...]
        <origin rpy="-1.57079633  0.64926248 -1.75545216" xyz="-0.02226000 -0.02707000 -0.11482000"/>
        <axis xyz="0.00000000 0.00000000 1.00000000"/>
        <limit lower="0.00000000" upper="2.00000000" effort="0" velocity="0"/>
    </joint>
    <joint name="j2s7s300_joint_finger_tip_1" type="fixed">
        <parent link="j2s7s300_link_finger_1"/>
        <child link="j2s7s300_link_finger_tip_1"/>
        <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.04400000 -0.00300000  0.00000000"/>
        <axis xyz="100"/>
    </joint>
    <joint name="j2s7s300_joint_finger_tip_2" type="fixed">
        <parent link="j2s7s300_link_finger_2"/>
        <child link="j2s7s300_link_finger_tip_2"/>
        <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.04400000 -0.00300000  0.00000000"/>
        <axis xyz="100"/>
    </joint>
    <joint name="j2s7s300_joint_finger_tip_3" type="fixed">
        <parent link="j2s7s300_link_finger_3"/>
        <child link="j2s7s300_link_finger_tip_3"/>
        <origin rpy="0.00000000 0.00000000 0.00000000" xyz="0.04400000 -0.00300000  0.00000000"/>
        <axis xyz="100"/>
    </joint>
</robot>
```
[kinovaJacoJ2S7S300.txt](https://github.com/user-attachments/files/16312758/kinovaJacoJ2S7S300.txt)

tmckenna-1, 2024-07-19:
Hi @martinleroux, thanks for the info. Just to clarify, is the urdf you supplied a corrected version, because adding up those masses I am only getting about 5.55 kg. It seems the difference between the inertial.xacro file and the one that you provided is the mass of link 7 (0.727 kg vs. 0.99 kg)

martinleroux, 2024-07-19:
5.55 kg matches the weight indicated in our User Guide (see page 12) and on our Website. I had not realized you were referring to the specsheet (I read your post too fast). At this time I don't know which value is the correct one - I will see if I can find a scale to put a robot on early next week and confirm whether the specsheet or the URDF is wrong.

Thank you for notifying us.

tmckenna-1, 2024-07-19:
@martinleroux, based on the spec sheet, 5.5 kg should be the weight of the arm without the 3 finger gripper, but your urdf includes the 3 finger gripper. The User Guide value on page 12 does not specify whether the listed weight includes the gripper or not (the picture associated does have the arm + gripper).
I weighed our arm without the gripper and got 5.5 kg. But let me know your findings once you have time to investigate further.

martinleroux, 2024-07-22:
I confirm, I just weighed an arm and got 5.5kg without the gripper. It's surprising that the data is wrong since we generate the URDF from CAD, but I'm clearly not going to argue with reality.

I'll escalate the issue with our engineering team, but unless they can find the issue in a matter of minutes, the investigation will likely only begin next sprint. If it is determined that we need to make actual measurements instead of relying on the cad, it may be a couple of months before we have public data.

In the meantime, the only thing I can offer is this snippet of code directly from the robot's firmware, but that is a piece of code nobody looked at for around 10 years so admittedly I don't know if the data is accurate (it does not sum to the right mass, but it does not include the base link).

```
// Links mass
			m_Gravity_Param_Mass[0] = 0.77;
			m_Gravity_Param_Mass[1] = 0.803;
			m_Gravity_Param_Mass[2] = 0.825;
			m_Gravity_Param_Mass[3] = 0.790;
			m_Gravity_Param_Mass[4] = 0.468;
			m_Gravity_Param_Mass[5] = 0.468;

			if (lowLevelManagement->deviceInspector.finger.GetCount() == 3)// If 3 fingers
			{
				m_Gravity_Param_Mass[6] = 1.025f;
			}
			else if (lowLevelManagement->deviceInspector.finger.GetCount() == 2)//If 2 fingers
			{
				m_Gravity_Param_Mass[6] = 0.910f;
			}
			else//If no hand
			{
				m_Gravity_Param_Mass[6] = 0.355f;
			}
```

tmckenna-1, 2024-07-23:
@martinleroux thanks for looking into this. Let me know what you can discover.

tmckenna-1, 2025-01-31:
Hello, pinging to see if there has been any update @martinleroux

martinleroux, 2025-01-31:
I had logged the request to engineering back when I said I did. The request was put up for triage in January (I admittedly don't know why it took so long). I just poked the program manager to get an update.